raspberrypi/picotool #4: No rule to install `picotool` binary

ggardet, 2021-01-29:
All the cmake system is in place except for the installation of `picotool` to `/usr/bin` target.
When `make install` is run, we get:
```
make: *** No rule to make target 'install'.  Stop.
```

kilograham, 2021-01-29:
good idea; that would make sense

aallan, 2021-02-01:
With respect to raspberrypi/pico-feedback#48 picotool should install into /usr/local/bin and not /usr/bin under macOS. Mucking about with /usr/bin on a Mac is problematic.

ggardet, 2021-02-01:
> With respect to [raspberrypi/pico-feedback#48](https://github.com/raspberrypi/pico-feedback/issues/48) picotool should install into /usr/local/bin and not /usr/bin under macOS. Mucking about with /usr/bin on a Mac is problematic.

It should be something `%{prefix}/bin` anyway

lurch, 2021-10-06:
Fixed by https://github.com/raspberrypi/picotool/commit/ce769705338cb57abc57d1a586983abe3a5e5199 ?

ggardet, 2021-10-13:
> Fixed by [ce76970](https://github.com/raspberrypi/picotool/commit/ce769705338cb57abc57d1a586983abe3a5e5199) ?

Likely, but I did not try it yet.

ggardet, 2021-10-20:
I backported ce76970 to version 1.0.1 and I confirm this fixes this issue.

closed 2021-10-20 by ggardet

lurch, 2021-10-20:
Why did you need to backport, instead of just using the latest `develop` version?