raspberrypi/picotool #8: How to convert a modified elf to uf2.

mytechnotalent, 2021-01-31:
I wanted to understand how to use the picotool to take a modified ```elf``` file and convert it to a ```uf2``` that can be dropped into the flash like a normal ```uf2```?

kilograham, 2021-01-31:
well if you really want to you can do
picotool load foo.elf
picotool save foo.uf2

but you are probably better of using elf2uf2 from the SDK

mytechnotalent, 2021-01-31:
@kilograham your amazing thank you for being so quick and so helpful!  I assume if I wanted to convert a uf2 to elf it would be:
```
picotool load foo.uf2
picotool save foo.elf
```

In addition is there a ```uf22elf``` as well in the SDK?  Thanks!

kilograham, 2021-01-31:
nope we don't have any tools that save as ELF

mytechnotalent, 2021-01-31:
Thanks again for all the help!  I was trying to find a way to work with the uf2 directly with radare2 but I believe it only works with elf.

mytechnotalent, 2021-01-31:
Is it ok I close this out?  You have been more than helpful!!!

kilograham, 2021-01-31:
i mean the build produces ELFs as well as UF2 ... i',m sure radare2 must work with BIN too? we can save that from picotool.

sure close away!

mytechnotalent, 2021-01-31:
@kilograham when I try:
```picotool load foo.uf2```
I get:
```No accessible RP2040 devices in BOOTSEL mode were found but, device at bus 1m address 16 appears to be a RP2040 device in BOOTSEL mode, but pictool was unable to connect```

In addition how would I flash a .bin (separate question).

kilograham, 2021-01-31:
if you are running on windows you need to install a driver INF. if you are on linux you probably need to be in the dialout group (or use sudo or use a udev rule).

Hopefully these are documented in the getting started book

closed 2021-01-31 by mytechnotalent