raspberrypi/picotool #13: Support "reset" command for non BOOTSEL device with STDIO USB reset interface

kilograham, 2021-03-06:
SDK 1.1.0 adds (optional) support for code exposing STDIO USB to be reset via another USB vendor interface. This obviously makes sense for the the "reset" command, though we should probably add a `-f, --force` or whatever option to make this happen.

lurch, 2021-03-08:
IIRC that new USB vendor interface allows resetting to BOOTSEL or resetting to flash boot, so perhaps picotool should expose both those options?

kilograham, 2021-03-08:
picotool already does

lurch, 2021-03-08:
I guess that must be only in the Graham-version? https://github.com/raspberrypi/picotool/commits/master

kilograham, 2021-03-08:
the `picotool reboot` already supports options to specify which type of boot you want (flash boot or usb boot), has since the get go; it is in master

lurch, 2021-03-08:
Ahh, I was getting confused by the already-existing-reboot (i.e. over PICOBOOT?) and the new-USB-vendor-reboot.
Also, I'd never thought to run `picotool help reboot` which is the **only** place where this is ever mentioned :stuck_out_tongue_winking_eye:

lurch, 2021-03-08:
And based on our internal discussions, I wonder if `picotool reboot` also ought to have an option for "reboot into BOOTSEL mode but only enable PICOBOOT, don't enable the MSD interface" ? :shrug:

closed 2023-03-05 by kilograham