raspberrypi/picotool #26: Support "list" command

lurch, 2021-04-29:
I wonder if a `pictool list` command might be useful? Consider the following scenario:
```
% lsusb | grep 2e8a                                       
Bus 020 Device 012: ID 2e8a:000a 2e8a Pico  Serial: 000000000000
Bus 020 Device 009: ID 2e8a:0005 2e8a Board in FS mode  Serial: 000000000000
% ./picotool info
No accessible RP2040 devices in BOOTSEL mode were found.

but:

Device at bus 20, address 9 appears to be a RP2040 MicroPython device not in BOOTSEL mode.
Device at bus 20, address 12 appears to be a RP2040 device with a USB serial connection, so consider -f or -F.
```
if I then plug in a Pico with a blank Flash chip (wiped using `flash_nuke.uf2`) I get:
```
% lsusb | grep 2e8a
Bus 020 Device 012: ID 2e8a:000a 2e8a Pico  Serial: 000000000000
Bus 020 Device 009: ID 2e8a:0005 2e8a Board in FS mode  Serial: 000000000000
Bus 020 Device 013: ID 2e8a:0003 2e8a RP2 Boot  Serial: E0C912D24340
% ./picotool info
Program Information
 none
```
In this latter case it would be nice if e.g. a `./picotool list` could tell me:
```
Device at bus 20, address 9 appears to be a RP2040 MicroPython device not in BOOTSEL mode.
Device at bus 20, address 12 appears to be a RP2040 device with a USB serial connection, so consider -f or -F.
Device at bus 20, address 13 appears to be a RP2040 device in BOOTSEL mode.
```
For bonus points, it could even include the other VID/PID combos from https://github.com/raspberrypi/usb-pid ? :wink:

lurch, 2021-07-19:
The absence of a `picotool list` command also makes it "tricky" to use the `picotool reboot` command (added by #23 ) when you've got two Picos plugged in 😉 

If **both** Picos are in BOOTSEL mode, then `picotool info` tells you e.g.
```
Multiple RP2040 devices in BOOTSEL mode found:

Device at bus 2, address 16:
----------------------------
Program Information
 name:      hello_usb
 web site:  https://github.com/raspberrypi/pico-examples/tree/HEAD/hello_world/usb
 features:  USB stdin / stdout

Device at bus 2, address 19:
----------------------------
Program Information
 name:      hello_usb
 web site:  https://github.com/raspberrypi/pico-examples/tree/HEAD/hello_world/usb
 features:  USB stdin / stdout
````
and if **both** Picos are in USB-CDC mode, then `picotool info` (in this example, on Windows) tells you:
```
No accessible RP2040 devices in BOOTSEL mode were found.

but:

Device at bus 2, address 20 appears to be a RP2040 device with a USB serial connection, not in BOOTSEL mode. You can force reset it into BOOTSEL mode via 'picotool reboot -f -u' first.
Device at bus 2, address 21 appears to be a RP2040 device with a USB serial connection, not in BOOTSEL mode. You can force reset it into BOOTSEL mode via 'picotool reboot -f -u' first.
```

However if one Pico is in USB-CDC mode and the other is in BOOTSEL mode, then `picotool info` (and any other picotool command?) seems to "only see" the device that's in BOOTSEL mode and just tells you:
```
Program Information
 name:      hello_usb
 web site:  https://github.com/raspberrypi/pico-examples/tree/HEAD/hello_world/usb
 features:  USB stdin / stdout
```

will-v-pi, 2026-11-05:
Merged into develop

closed 2026-11-05 by will-v-pi