raspberrypi/picotool #29: Make picotool build on FreeBSD

arg08, 2021-05-15:
Picotool currently doesn't quite build on FreeBSD because libusb (which is part of the base system rather than an add-on package) is just 'libusb' rather than 'libusb-1.0' even though it's compatible with the Linux -1.0 version.

So very simple patch makes it all work:

diff --git a/cmake/FindLIBUSB.cmake b/cmake/FindLIBUSB.cmake
index c8cc62c..560e023 100644
--- a/cmake/FindLIBUSB.cmake
+++ b/cmake/FindLIBUSB.cmake
@@ -20,7 +20,7 @@ else (LIBUSB_INCLUDE_DIR AND LIBUSB_LIBRARIES)
     ENDIF(NOT WIN32)
     FIND_PATH(LIBUSB_INCLUDE_DIR libusb.h
             PATHS ${PC_LIBUSB_INCLUDEDIR} ${PC_LIBUSB_INCLUDE_DIRS})
-    FIND_LIBRARY(LIBUSB_LIBRARIES NAMES usb-1.0
+    FIND_LIBRARY(LIBUSB_LIBRARIES NAMES usb-1.0 usb
             PATHS ${PC_LIBUSB_LIBDIR} ${PC_LIBUSB_LIBRARY_DIRS})
     include(FindPackageHandleStandardArgs)
     FIND_PACKAGE_HANDLE_STANDARD_ARGS(LIBUSB DEFAULT_MSG LIBUSB_LIBRARIES LIBUSB_INCLUDE_DIR)

ie. just add "usb" after "usb-1.0" on the FIND_LIBRARY line.

aallan, 2021-05-15:
See also pico-sdk#396.

lurch, 2021-05-15:
Thanks, I've incorporated this suggestion into #27 :+1:

lurch, 2021-10-05:
Fixed by https://github.com/raspberrypi/picotool/commit/757f2aee3c2c988b4d0c9c15907e243350381d0a ?

closed 2023-02-10 by kilograham

faelpinho, 2023-10-26:
Just for note: I could build and install on Manjaro just forcing include to libusb.
More: https://faelpinho.medium.com/how-to-install-picotool-on-manjaro-linux-6a5a3cfaf743