raspberrypi/picotool #32: add a udev rule for linux systems to access the pico without sudo

diehard67, 2021-05-27:
I created a udev rules file so I can use picotool without sudo

```
# make raspberry pi pico accessible without sudo.

SUBSYSTEM!="usb_device", ACTION!="add", GOTO="rpi2_end"
# Raspberry Pi Pico
ATTR{idVendor}=="2e8a", ATTRS{idProduct}=="0003", MODE="660", GROUP="plugdev"

LABEL="rpi2_end"
```

lurch, 2021-05-27:
I guess this is very similar to https://github.com/raspberrypi/openocd/pull/5 and https://github.com/raspberrypi/openocd/pull/27 ?
(although not identical, as it has a different `idProduct`)

diehard67, 2021-05-28:
yah sort of, this one is for a pico in BOOTSEL mode, not picoprobe, but looks like the same goal.

kilograham, 2023-02-10:
merged into develop

closed 2023-02-10 by kilograham