raspberrypi/picotool #34: Build errors with GCC11

ggardet, 2021-06-03:
I have the following build errors with GCC11:
```
[   19s] [ 50%] Building CXX object CMakeFiles/picotool.dir/main.cpp.o
[   19s] /usr/bin/c++ -DPICO_BUILD=1 -DPICO_NO_HARDWARE=1 -DPICO_ON_DEVICE=0 -I/usr/include/libusb-1.0 -I/home/abuild/rpmbuild/BUILD/picotool-1.0.1/pico-sdk-1.1.2/src/common/pico_binary_info/include -I/home/abuild/rpmbuild/BUILD/picotool-1.0.1/pico-sdk-1.1.2/src/common/boot_uf2/include -I/home/abuild/rpmbuild/BUILD/picotool-1.0.1/pico-sdk-1.1.2/src/common/boot_picoboot/include -I/home/abuild/rpmbuild/BUILD/picotool-1.0.1/pico-sdk-1.1.2/src/host/pico_platform/include -I/home/abuild/rpmbuild/BUILD/picotool-1.0.1/picoboot_connection -O2 -Wall -D_FORTIFY_SOURCE=2 -fstack-protector-strong -funwind-tables -fasynchronous-unwind-tables -fstack-clash-protection -Werror=return-type -flto=auto -g -DNDEBUG -O2 -g -DNDEBUG -std=gnu++14 -MD -MT CMakeFiles/picotool.dir/main.cpp.o -MF CMakeFiles/picotool.dir/main.cpp.o.d -o CMakeFiles/picotool.dir/main.cpp.o -c /home/abuild/rpmbuild/BUILD/picotool-1.0.1/main.cpp
[   19s] In file included from /home/abuild/rpmbuild/BUILD/picotool-1.0.1/main.cpp:11:
[   19s] /home/abuild/rpmbuild/BUILD/picotool-1.0.1/cli.h: In member function 'D& cli::matchable_derived<T>::repeatable()':
[   19s] /home/abuild/rpmbuild/BUILD/picotool-1.0.1/cli.h:257:25: error: 'numeric_limits' is not a member of 'std'
[   19s]   257 |             _max = std::numeric_limits<int>::max();
[   19s]       |                         ^~~~~~~~~~~~~~
[   19s] /home/abuild/rpmbuild/BUILD/picotool-1.0.1/cli.h:257:40: error: expected primary-expression before 'int'
[   19s]   257 |             _max = std::numeric_limits<int>::max();
[   19s]       |                                        ^~~
[   19s] /home/abuild/rpmbuild/BUILD/picotool-1.0.1/cli.h: At global scope:
[   19s] /home/abuild/rpmbuild/BUILD/picotool-1.0.1/cli.h:444:31: error: 'numeric_limits' is not a member of 'std'
[   19s]   444 |         int _max_value = std::numeric_limits<int>::max();
[   19s]       |                               ^~~~~~~~~~~~~~
[   19s] /home/abuild/rpmbuild/BUILD/picotool-1.0.1/cli.h:444:46: error: expected primary-expression before 'int'
[   19s]   444 |         int _max_value = std::numeric_limits<int>::max();
[   19s]       |                                              ^~~
[   19s] /home/abuild/rpmbuild/BUILD/picotool-1.0.1/cli.h: In lambda function:
[   19s] /home/abuild/rpmbuild/BUILD/picotool-1.0.1/cli.h:408:36: error: 'numeric_limits' is not a member of 'std'
[   19s]   408 |                 long lvalue = std::numeric_limits<long>::max();
[   19s]       |                                    ^~~~~~~~~~~~~~
[   19s] /home/abuild/rpmbuild/BUILD/picotool-1.0.1/cli.h:408:51: error: expected primary-expression before 'long'
[   19s]   408 |                 long lvalue = std::numeric_limits<long>::max();
[   19s]       |                                                   ^~~~
[   19s] /home/abuild/rpmbuild/BUILD/picotool-1.0.1/cli.h: At global scope:
[   19s] /home/abuild/rpmbuild/BUILD/picotool-1.0.1/cli.h:500:40: error: 'numeric_limits' is not a member of 'std'
[   19s]   500 |         unsigned int _max_value = std::numeric_limits<unsigned int>::max();
[   19s]       |                                        ^~~~~~~~~~~~~~
[   19s] /home/abuild/rpmbuild/BUILD/picotool-1.0.1/cli.h:500:55: error: expected primary-expression before 'unsigned'
[   19s]   500 |         unsigned int _max_value = std::numeric_limits<unsigned int>::max();
[   19s]       |                                                       ^~~~~~~~
[   19s] /home/abuild/rpmbuild/BUILD/picotool-1.0.1/cli.h: In lambda function:
[   19s] /home/abuild/rpmbuild/BUILD/picotool-1.0.1/cli.h:460:36: error: 'numeric_limits' is not a member of 'std'
[   19s]   460 |                 long lvalue = std::numeric_limits<long>::max();
[   19s]       |                                    ^~~~~~~~~~~~~~
[   19s] /home/abuild/rpmbuild/BUILD/picotool-1.0.1/cli.h:460:51: error: expected primary-expression before 'long'
[   19s]   460 |                 long lvalue = std::numeric_limits<long>::max();
[   19s]       |                                                   ^~~~
```

lurch, 2021-06-03:
I think this is fixed by #30 ?

ggardet, 2021-06-03:
> I think this is fixed by #30 ?

Indeed.

closed 2021-06-24 by kilograham