raspberrypi/picotool #38: Device in BOOTSEL cant be seen

gusisin, 2021-08-08:
My device mounts in BOOTSEL and shows the file system but pico tool reports 
`pi@raspberrypi:~/pico/picotool $ picotool info --bus 1 --address 21
No accessible RP2040 device in BOOTSEL mode was found at bus 1, address 21.

but:

Device at bus 1, address 21 appears to be a RP2040 device in BOOTSEL mode, but picotool was unable to connect`

environment:
Built following getting started guide
Linux raspberrypi 5.10.52-v7l+ #1441 SMP Tue Aug 3 18:11:56 BST 2021 armv7l GNU/Linux
 pico connected via USB

lurch, 2021-08-09:
Have you tried using `sudo` ?

closed 2023-02-10 by kilograham

Sukruthi-C, 2023-12-26:
How did you fix this?

lurch, 2023-12-26:
@Sukruthi-C See the section in https://github.com/raspberrypi/picotool/blob/master/README.md about udev rules. Alternatively, run picotool with sudo, as I mentioned above.

danergo, 2024-03-12:
Resurrecting this thread: QT Py RP2040: Windows11, PIO, picotool sees Nothing.

lurch, 2024-03-12:
@danergo You should probably ask for help on the Adafruit forums.

danergo, 2024-03-13:
Thank you! It turned out this was caused by a bad driver. :) Correct one is "WinUSB (v6.1.7600.16385)" as of now.