raspberrypi/picotool #39: picotool crashes when given an ELF build with embedded rust

jonathanpallant, 2021-10-05:
```console
$ cargo build --example pico_blinky
   Compiling pico v0.1.0 (/home/jonathan/rp-hal/boards/pico)
    Finished dev [unoptimized + debuginfo] target(s) in 0.91s
$ ../picotool/build/picotool info ./target/thumbv6m-none-eabi/debug/examples/pico_blinky
ERROR: filename './target/thumbv6m-none-eabi/debug/examples/pico_blinky' does not have a recognized file type (extension)
$ cp ./target/thumbv6m-none-eabi/debug/examples/pico_blinky ./pico_blinky.elf
$ ../picotool/build/picotool info ./pico_blinky.elf
picotool: /home/jonathan/picotool/main.cpp:156: void range_map<T>::check_overlap(uint32_t) [with T = long unsigned int; uint32_t = unsigned int]: Assertion `p >= f->first' failed.
[1]    3871 abort      ../picotool/build/picotool info ./pico_blinky.elf
```

lurch, 2021-10-05:
If it's asserting in `check_overlap` could you add some extra debug code to picotool to see which sections it thinks are overlapping? I know almost nothing about ELF myself, but I suspect @kilograham will be able to investigate this easier if you can provide a copy of your built `pico_blinky.elf` ? :slightly_smiling_face:

jonathanpallant, 2021-10-05:
I think it's an ordering issue. You always check against the lowest item in the map.

jonathanpallant, 2021-10-05:
[pico_blinky.zip](https://github.com/raspberrypi/picotool/files/7285084/pico_blinky.zip)

jonathanpallant, 2021-10-05:
I can confirm that sorting our sections fixes it the crash.
[pico_blinky_fixed.zip](https://github.com/raspberrypi/picotool/files/7285195/pico_blinky_fixed.zip)

kilograham, 2021-10-05:
yup this is a bug in the checking code, you can comment it out for now as a workaround - will fix it pretty sooin

jannic, 2022-06-05:
`picotool` 1.1.0 now does write ELF binaries built with embedded rust without crashing, but the resulting flash contents are broken.

This is caused by the ELF binaries containing non-contiguous ranges which end / begin in the same flash erase page. Eg.:
```
Section Headers:
  [Nr] Name              Type            Addr     Off    Size   ES Flg Lk Inf Al
  [ 0]                   NULL            00000000 000000 000000 00      0   0  0
  [ 1] .vector_table     PROGBITS        10000100 010100 0000a8 00   A  0   0  4
  [ 2] .boot2            PROGBITS        10000000 020000 000100 00   A  0   0  1
  [ 3] .text             PROGBITS        100001a8 0201a8 00ec8c 00  AX  0   0  4
  [ 4] .rodata           PROGBITS        1000ee40 02ee40 00185c 00  AM  0   0 16
  [ 5] .data             PROGBITS        20000000 03069c 000000 00   A  0   0  4
  [ 6] .gnu.sgstubs      PROGBITS        100106a0 0306a0 000000 00   A  0   0 32
  [ 7] .bss              NOBITS          20000000 040000 000004 00  WA  0   0  4
  [ 8] .uninit           NOBITS          20000004 040000 000000 00  WA  0   0  4
```

Adding some diagnostic output printing the range_map entries shows:
```
Found memory range 0x10000100->0x100001a8
Found memory range 0x10000000->0x10000100
Found memory range 0x100001a8->0x1000ee34
Found memory range 0x1000ee40->0x1001069c
```
Notice that there's a hole between `0x1000ee34` and `0x1000ee40`.

When writing the last range, the code at https://github.com/raspberrypi/picotool/blob/master/main.cpp#L1854-L1860 will zero the whole erase block, causing the flash contents from `0x1000e000` to `0x1000ee34` to be overwritten with zeros.

This matches what I can see after dumping the resulting flash contents.

lurch, 2022-06-05:
> This is caused by the ELF binaries containing non-contiguous ranges

Possibly a similar problem to https://github.com/raspberrypi/pico-sdk/issues/762 ?

> the code at ... will zero the whole erase block

Hmmm, perhaps picotool contains a similar bug to [pico-bootrom](https://github.com/raspberrypi/pico-bootrom/issues/19) ?

jannic, 2022-11-19:
Just as a note: I wondered why the ELF files had those gaps between sections at all.
Found this questions which might be related: https://stackoverflow.com/questions/66644603/gnu-ld-for-arm-produces-section-alignment-to-unwanted-bound
TL;DR: It looks like some linker versions increase the alignment of some sections, which can cause such gaps.

kilograham, 2023-02-09:
@jannic any chance you can attach a UF2 with holes

jannic, 2023-02-09:
> @jannic any chance you can attach an ELF/UF2 with holes

Sure:
[pico_blinky.zip](https://github.com/raspberrypi/picotool/files/10697097/pico_blinky.zip)
(Zipped because github doesn't like raw elf files)

Flashing that file `picotool load pico_blinky -t elf` fails with 1.1.0 and works with #51.
(The flash command itself seems to succeed, but the flash contents are wrong and the blinky example doesn't work.)

`readelf -S` of that binary:
```
Section Headers:
  [Nr] Name              Type            Addr     Off    Size   ES Flg Lk Inf Al
  [ 0]                   NULL            00000000 000000 000000 00      0   0  0
  [ 1] .vector_table     PROGBITS        10000100 000134 0000a8 00   A  0   0  4
  [ 2] .boot2            PROGBITS        10000000 0001dc 000100 00   A  0   0  1
  [ 3] .text             PROGBITS        100001a8 0002dc 00d974 00  AX  0   0  4
  [ 4] .rodata           PROGBITS        1000db20 00dc50 0012b0 00  AM  0   0 16
  [ 5] .data             PROGBITS        20000000 00ef00 000000 00   A  0   0  4
[...]
```

kilograham, 2023-02-10:
merged into develop

closed 2023-02-10 by kilograham