raspberrypi/picotool #48: Unable to see device via --address flag on Windows

neilenns, 2022-01-23:
Steps to reproduce:

1. Plug in a RP2040 Pico to Windows 11 while holding down the BOOTSEL button 
2. Type `picotool info`

Result: device is found, reports no program information (correct)

3. Type `picotool info --address 5`

Result: "No accessible RP2040 devices in BOOTSEL mode were found with address 5."
Expected result: device is found and reports no program information

I'm using WinUSB v6.1.760.16385 for the driver, installed via zadig. The device address comes from the device properties in Device Manager.

neilenns, 2022-01-24:
For reference here's what the address looks like in device manager:

[image]

Taking a peek at the code it looks like it automatically gets converted to an `int`. Maybe the issue on Windows is the address isn't actually `5` it's `00000005`?

neilenns, 2022-01-24:
I've also tried with `libusb-win32 (v1.2.6.0)` just for fun but no change in behaviour.

lurch, 2022-01-24:
See also #26

neilenns, 2022-01-24:
@lurch I did see that as well. In my case I am trying to programmatically run picotool so I need some way to identify the boards beforehand then pass in the address of the board I want to reference.

For Arduinos this is done with the COM port, and that port is available in the Windows registry. I can enumerate the USB devices from the registry, find the COM port, then call `avrdude` accordingly.

I'm struggling to do the same with picotool because I can't find the address picotool is using in the registry (which shows `5` not `15` and a bus of `0` which picotool also doesn't think is valid).

lurch, 2022-01-24:
Perhaps picotool is using some kind of "libusb addressing scheme" which is different to the Device-Manager addressing scheme? :shrug:  (the fun thing about USB is how many different "layers" there are!)

And just in case it's not obvious - AFAIK it's _only_ necessary to use the `--address` flag if you've got more than one Pico connected at once. With only a single Pico in use, `picotool` should just always correctly auto-detect it.

neilenns, 2022-01-24:
@lurch In my situation it's possible there will be many Picos connected at the same time so I need to be able to directly address them.

I agree, I think there's some sort of magic libusb addressing scheme going on. Unfortunately that's yet another layer down in the libusb code so I'm not sure how it's doing it :(

neilenns, 2022-01-24:
I was able to get a sample app running using a .NET USB library (which I can integrate into my project) and enumerate the devices and see the address (15) and bus (2).

Closing this since it's obviously not a picotool problem.

closed 2022-01-24 by neilenns