raspberrypi/picotool #49: load command fails when specifying bus and address

neilenns, 2022-01-25:
I'm unable to flash a device using picotool with a specified bus and address even though the same device can be seen using the info command. This is on Windows 11 with the libusb driver installed via zadig. The device is in BOOTSEL mode.

When I run `.\picotool.exe info -a --bus 2 --address 12` I get back a response showing the flash size and rom version.

When I run `.\picotool.exe load firmware.uf2` the firmware loads successfully.

When I run `.\picotool.exe load firmware.uf2 --bus 2 --address 12` I get back "No accessible RP2040 device in BOOTSEL mode was found at bus 2, address 12."

Why doesn't the load command work with the bus and address specified?

neilenns, 2022-01-25:
The reboot command works fine as well: `.\picotool.exe reboot --bus 2 --address 12` reboots the device back into application mode.

lurch, 2022-01-25:
I dunno if libusb works the same on Windows, but at least on Linux the RP2040 stays on the same "bus" but gets an incrementing "address" number each time I unplug and replug it (from the same USB port) - so maybe what _was_ `--bus 2 --address 12` is now `--bus 2 --address 13` ? :shrug:

neilenns, 2022-01-25:
@lurch All the above commands were run in sequence without reconnecting the device.

... and of course as I went to reproduce this it just worked. Sigh.

I'll see if I can reproduce this again and if I can't I'll close it tomorrow.

neilenns, 2022-01-26:
Ok this is wacky, but it appears to have something to do with how I specify the path to picotool. Crazy. I'm in this folder: `D:\git\MobiFlight-Connector\RaspberryPi\tools`.

If I run this command it works:

` .\picotool.exe info --bus 2 --address 12`

If I run this command it fails:

`D:\git\MobiFlight-Connector\bin\Debug\RaspberryPi\tools\picotool.exe info --bus 2 --address 12`

[image]

This makes no sense at all 😂

neilenns, 2022-01-26:
Oh man, what an adventure. I got it figured out, I had two versions of picotool in two different folders (one I got from who knows where and one I built locally). The one I built locally worked fine, the other didn't.

Once I got that sorted things worked. Man, what a mess! Oops :)

closed 2022-01-26 by neilenns