raspberrypi/picotool #53: "LibUSB is not found" error when PkgConfig is missing

Pwuts, 2022-07-06:
I just installed LibUSB, then ran `PICO_SDK_PATH=[path_to_sdk] cmake ..` just like the README says. I then received the following error:
```
Using PICO_SDK_PATH from environment ('/home/reinier/code/badgeteam/rpi-pico-sdk')
-- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE)
-- Could NOT find LIBUSB (missing: LIBUSB_INCLUDE_DIR)
CMake Error at CMakeLists.txt:34 (message):
  picotool cannot be built because libUSB is not found
```
I then ran `sudo apt install pkg-config` after which the build succeeded. This makes me think the error I originally received is erroneous. :)

Pwuts, 2022-07-09:
Yes, I should have followed the instructions completely and in the right order, but when it's PkgConfig that is missing it shouldn't complain about LibUSB while that one actually is available. :)

lurch, 2022-07-09:
I just dug into this a bit... the reason that it's complaining that it can't find libUSB (even though it's installed), is because it uses pkg-config itself to find libUSB.

I'm no CMake expert, but it It looks like the "Could NOT find PkgConfig" message is coming from https://github.com/raspberrypi/picotool/blob/master/cmake/FindLIBUSB.cmake#L18 , the "Could NOT find LIBUSB" message is coming from https://github.com/raspberrypi/picotool/blob/master/CMakeLists.txt#L32 , and the "picotool cannot be built because libUSB is not found" message is coming from https://github.com/raspberrypi/picotool/blob/master/CMakeLists.txt#L34 .
@kilograham Should there be a fatal error if https://github.com/raspberrypi/picotool/blob/master/cmake/FindLIBUSB.cmake#L18 isn't able to find the PkgConfig module? :shrug:

NickAlvar91, 2022-08-24:
Were you able to fix your stuff?
[image]

I had that EXACT problem when building no mater what i used. LibUSB changed the folders where the libraries are installed so CMAKE gets upset 
[image]

if you correct the paths to point at libusb.h and libusb-1.0.lib. that should fix atleast that part. 

I give a little more of a detailed explanation in the issue  linked below. 
https://github.com/raspberrypi/picotool/issues/52

if you follow the details linked below it should also help (assuming you make sure to grab "libusb-1.0.24")
https://shawnhymel.com/2168/how-to-build-openocd-and-picotool-for-the-raspberry-pi-pico-on-windows/

lurch, 2022-08-26:
> @Pwuts Were you able to fix your stuff?

@NickAlvar91 The original message says "after which the build succeeded" so I think this is actually a slightly different problem to #52 .  Also, the OP is using Linux (`sudo apt install ...`) whereas you're using Windows? :wink:

will-v-pi, 2024-08-09:
This should all be fixed with picotool 2.0.0 if you download the latest libusb 1.0.27 release on Windows, or follow the build instructions on Linux / macOS. We won't be adding a way to find libusb without pkg-config on Linux / macOS, as pkg-config installation is relatively simple

closed 2024-08-09 by will-v-pi