raspberrypi/picotool #54: Feature: Target devices by it's unique ID

ggsubs, 2022-07-30:
Feature request: Target devices by their unique ID, so that I can work with multiple devices connected at the same time, while allowing the USB port connections to change.

The unique id is available on the SDK (Target devices by it's unique ID), the same string is exposed on the device descriptor.

ggsubs, 2022-07-30:
Example: slusb -v output shows iSerial value:

```
Bus 002 Device 004: ID 2e8a:000a  
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x2e8a 
  idProduct          0x000a 
  bcdDevice            1.00
  iManufacturer           1 Raspberry Pi
  iProduct                2 Pico
  iSerial                 3 E66038B713249237
  bNumConfigurations      1
```

neilenns, 2022-08-19:
Yes, this would be excellent. I opened similar issues a while back (https://github.com/raspberrypi/picotool/issues/48) and despite how I closed that issue I actually never solved my problem. I couldn't find any way to programmatically reference an RP2040 device and flash it using picotool.

lurch, 2023-03-05:
I guess this ties up closely with #12 (since the ID returned by `pico_get_unique_board_id` **is** the flash-chip's unique ID.

Ferdi265, 2023-04-25:
I have created a very experimental branch [here](https://github.com/ferdi265/picotool/tree/rewrite-main) that provides a `--serial` option for selecting devices, as well as `--multiple` for e.g., rebooting multiple devices into BOOTSEL mode at once.

The branch is pretty experimental, but it simplifies the workflow for working with multiple picos by a lot.

For example:
```
picotool load server.uf2 --serial $SERVER_SERIAL -f
picotool load client.uf2 --serial $CLIENT_SERIAL -f
```
will correctly reboot two specific picos into BOOTSEL mode, flash them, and reboot them back into application mode.
I'm still cleaning up the branch, but I hope I will have something quite usable in a week or two.

will-v-pi, 2024-08-09:
This has been added with picotool 2.0.0 as the `--ser` option

closed 2024-08-09 by will-v-pi