raspberrypi/picotool #56: --force doesn't work on MacOS

playduck, 2022-08-27:
Executing any command with `-f` or `-F` has no effect.

picotool either responds with "No accessible RP2040 devices in BOOTSEL mode were found." or, if the device is manually put into BOOTSEL mode, it flashes the device.
Using `-f`, I'd expect the device to be rebooted. This is not the case.
Everything else regarding picotool seems to work as expected.

I'm using running on an M1 using:

- cmake version 3.24.1
- Apple clang version 13.1.6 (clang-1316.0.21.2.5) Target: arm64-apple-darwin21.6.0
- libusb-1.0, version 1.0.26
- picotool v1.1.0 (Darwin 21.6.0, AppleClang-13.1.6.13160021, Release)
- picotool pulled at `03f28122cc170535f35e5bf8fa37be024489f5eb`
- pico-sdk pulled at `2e6142b15b8a75c1227dd3edbe839193b2bf9041`

The pico is connected via a USB A to C adapter (also tested with a USB hub yielding no difference).

The cmake and make output is listed here:
```bash
➜  build git:(master) cmake ..
-- The C compiler identification is AppleClang 13.1.6.13160021
-- The CXX compiler identification is AppleClang 13.1.6.13160021
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
Using PICO_SDK_PATH from environment ('/Users/robin/Documents/code/pico/pico-sdk')
-- Found PkgConfig: /opt/homebrew/bin/pkg-config (found version "0.29.2")
-- Checking for module 'libusb-1.0'
--   Found libusb-1.0, version 1.0.26
-- Found LIBUSB: /opt/homebrew/lib/libusb-1.0.dylib
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/robin/Documents/code/pico/picotool/build
➜  build git:(master) ✗ make
[ 25%] Building CXX object CMakeFiles/picotool.dir/main.cpp.o
[ 50%] Building CXX object CMakeFiles/picotool.dir/picoboot_connection/picoboot_connection_cxx.cpp.o
[ 75%] Building C object CMakeFiles/picotool.dir/picoboot_connection/picoboot_connection.c.o
[100%] Linking CXX executable picotool
[100%] Built target picotool
```

I'm not sure wether this is a bug in picotool or a problem on my setup.

lurch, 2022-08-30:
IIRC the 'force' options only work when the program running on your Pico is using USB-CDC (i.e. USB stdio), as it uses [PICO_STDIO_USB_ENABLE_RESET_VIA_VENDOR_INTERFACE](https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/pico_stdio_usb/include/pico/stdio_usb.h#L80)

playduck, 2022-08-30:
Thanks!
Setting `pico_enable_stdio_usb(name 1)` in the `CMakeLists.txt` worked.
I've probably missed that in the documentation.

closed 2022-08-30 by playduck

mark2b, 2022-08-30:
I have the same problem, where I need to add pico_enable_stdio_usb(name 1) ?
CMakeLists.txt of picotool  ?
Can you attach an example ?
Sorry for newbie question

kilograham, 2022-08-30:
TLDR for why; by default when you are running your own application on the Pico, there is no USB interface for `picotool` to talk to. If your app includes stdio over USB using the SDK, we (by default) add another interface that `picotool` can use to reset the board back into BOOTSEL mode.

playduck, 2022-08-30:
@mark2b I've enabled usb in [my projects' CMakeLists.txt](https://github.com/playduck/pico-dsp/blob/master/CMakeLists.txt#L42) (and disabled uart in the line above).
Then I'm enabling stdio in [my main function](https://github.com/playduck/pico-dsp/blob/master/main.cpp#L50) (not sure if strictly required but helpful to enable printf nonetheless).

mark2b, 2022-08-30:
@kilograham @playduck Thanks a lot
BTW enabling stdio_init_all(); in additional to CMakeLists solved my problem
I use script to redeploy binary w/o manually switch to BOOTSEL

lurch, 2022-08-31:
> If your app includes stdio over USB using the SDK, we (by default) add another interface that picotool can use to reset the board back into BOOTSEL mode.

@kilograham Just out of curiosity, is there (or could there be) a way to add the picotool-reset interface to an app _without_ also enabling stdio over USB? Or is that a nonsensical question? (I dunno much about USB!)

kilograham, 2022-08-31:
> @kilograham Just out of curiosity, is there (or could there be) a way to add the picotool-reset interface to an app _without_ also enabling stdio over USB? Or is that a nonsensical question? (I [...]

You need to have a USB device stack running, and a USB interface which matches the one we provide along side stdio USB (it isn't particular to stdio_usb, that is just the only thing in the SDK which enables the USB device stack). You can create your own reset interface alongside your own use of TinyUSB