raspberrypi/picotool #58: Add support for read/writing files on embedded drives

kilograham, 2022-11-11:
picotool is able to find embedded drives; it might be nice to be able to list/extract/and writes files

will-v-pi, 2024-11-28:
I've got this working for LittleFS (which is used by MicroPython) on the [block-device-support](https://github.com/raspberrypi/picotool/tree/block-device-support) branch - so far I've added `ls` (listing all files/directories recursively), `mkdir`, `put` and `get`. I'll add `mv` and `rm` too - are there any other commands that are worth adding, and also what other filesystems should this support?

lurch, 2024-11-28:
> are there any other commands that are worth adding?

Perhaps https://docs.micropython.org/en/latest/reference/mpremote.html#mpremote-command-fs suggests the kind of commands that might be useful? :slightly_smiling_face: 

> also what other filesystems should this support?

FAT seems like the obvious suggestion?

lurch, 2024-11-28:
Just in case the GH notifications don't work properly: I've left some comments at the bottom of https://github.com/raspberrypi/picotool/commit/626beaa146f44cb26e9ed27f094fc1c41b1a93d4

kilograham, 2024-11-28:
Note (if you don't already) we should probably allow use of a partition as well as the embedded drive (since people will probably use those in the future)

lurch, 2024-11-29:
> Note (if you don't already) we should probably allow use of a partition as well as the embedded drive (since people will probably use those in the future)

Oh, they're different things? Just goes to show how little I understand all of this :rofl: