raspberrypi/picotool #63: Getting picotool on Linux?

seamusdemora, 2022-12-20:
I've located my Pico C SDK development platform on my Ubuntu 22.04. I'm worried that I'll wear out my USB port with all of the plugging and unplugging. I understand that `picotool` offers a solution for this - allowing one to build & upload without the plugging, un-plugging & button-pushing (a la Arduino's IDE).  I've looked in at the README here & Appendix B in the [Getting Started Guide](https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf), but I am still unclear on how to get `picotool` installed. Have I missed that?? 

Here's my directory setup... if someone could tell me where I need to insert the required directory, I think I can go from there. 
```bash
~/pico$ tree -L 1
.
├── ina260_i2c
├── MyProjects
├── pico-examples
├── pico-extras
├── pico-playground
├── pico-sdk
```

JamesH65, 2022-12-20:
Please ask this on the forums, in the meantime, read Appendix B of this document. https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf

seamusdemora, 2022-12-22:
@JamesH65 : 

I can't use the forums because I'm "banned for life"!  An exchange with someone on the forum a few years ago resulted in a lifetime ban. More recently, Elwell told me he checked & I was not banned. So I went back to the forums until the "gatekeepers" discovered me there, and double-banned me - they accused me of trying to "cheat" their original ban (not true!!). Therefore, I don't think I'm welcome there at all. Perhaps not here, either?  If you want to check into this on my behalf, please feel free.

Anyway - the documentation I read said that PicoTool would allow me to upload my files on the Pico
   * without the extra cabling need for SWD 
   * without wearing my USB connector out 
Did I mis-read that also? Will `picotool` not do this??

And as I said, I did read Appendix B in the [Getting Started Guide](https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf), but it fell a wee bit short of helping me. I think all I need to know is what directory to begin the process outlined in Appendix B - a relative directory location is given (../) instead of a fixed one.

lurch, 2022-12-22:
> the documentation I read said that PicoTool would allow me to upload my files on the Pico
> * without the extra cabling need for SWD

It will, **if** every UF2 you run on your Pico uses USB-STDIO, as that enables an extra USB end-point that picotool uses to flash binaries to the Pico and/or reset it. (at least that's the gist of things, I might not have got the terminology 100% correct)
As soon as you flash a UF2 that doesn't use USB-STDIO, this extra USB interface that picotool uses will no longer exist.

> without wearing my USB connector out

If you're worried about wearing out your USB connector, and you don't want to use SWD, have a look at https://www.raspberrypi.com/news/how-to-add-a-reset-button-to-your-raspberry-pi-pico/

seamusdemora, 2022-12-23:
@lurch 

Thanks - but can you tell me which directory directory I should use to install PicoTool? The instructions in the documentation seem confusing - but perhaps I'm dense - or missing something. To be clear: 
```
Getting picotool
The picotool utility is available in its own repository. You will need to clone and build it if you haven’t ran the pico-setup
script.
$ git clone https://github.com/raspberrypi/picotool.git --branch master
$ cd picotool
```   
### Question: what directory should I be in when I run `git clone`? That determines where the repo will be located, and it would seem that it must be in the proper location to work. Am I wrong?  

```
Building picotool can be done as follows, 

$ mkdir build
$ cd build
$ export PICO_SDK_PATH=~/pico/pico-sdk $ cmake ../
$ make
```  

This is confusing me because from my `~/pico` dir (where everything "pico-related" is housed there is already a `build` directory. A `tree` of `~/pico` is as follows, and `build` is under `pico-examples`: 
```
~/pico$ tree -L 1
.
├── ina260_i2c
├── MyProjects
├── pico-examples   <=== a `build` directory is under pico-examples
├── pico-extras
├── pico-playground
├── pico-sdk
```

If I can get the required directory location from which to `clone` the PicoTool repo, and where the `build` directory must be located, I think I can manage it from there. Again, I'm using an Ubuntu box as my development machine, so I wasn't able to run the RPi installer script. 

Thanks!

lurch, 2022-12-24:
> Question: what directory should I be in when I run git clone?

It doesn't matter.

> and build is under pico-examples

That also doesn't matter. You typically don't have a single `build` directory, but one `build` directory per repo (or sometimes multiple `build` directories per repo, if you're building projects individually rather than all-at-once).

seamusdemora, 2022-12-24:
Ah! Thank you muchly & Merry Xmas!

closed 2022-12-24 by seamusdemora

lurch, 2022-12-24:
:christmas_tree: