raspberrypi/picotool #64: picoprobe load works, then does not

d-adamson, 2023-01-26:


d-adamson, 2023-01-27:
Sorry, I wrote "picoprobe" in the title, but meant "picotool" (was a little tired at the time). Updated the title.
@JamesH65. Not necessarily. Replicated the behavior using the example from the sdk pico_w/blink

**FIRST ATTEMPT - OK**
~/pico/pico-examples/build/pico_w/blink$ picotool load -f picow_blink.uf2
The device was asked to reboot into BOOTSEL mode so the command can be executed.

Loading into Flash: [==============================]  100%

The device was asked to reboot back into application mode.

**SECOND ATTEMPT - NO GO**
~/pico/pico-examples/build/pico_w/blink$ picotool load -f picow_blink.uf2
No accessible RP2040 devices in BOOTSEL mode were found.

lurch, 2023-01-27:
As it notes in the [picotool README](https://github.com/raspberrypi/picotool/blob/master/README.md) "...but running compatible code..." i.e. there needs to be "something listening" on the Pico-side of the USB connection for these picotool commands. Which is typically either BOOTSEL mode or USB stdio. And so if you've not enabled USB stdio in your application, and you've "rebooted into application mode" (i.e. your application is actually running) then there's nothing on your Pico listening for picotool commands. Which perfectly explains the behaviour you're seeing.

P.S. If your Pico _is_ in BOOTSEL mode, then neither of the `-f / -F` flags are necessary.

d-adamson, 2023-01-27:
Thanks!
Yeah, I wasn't sure what was meant by "...but running compatible code..."
Seems to be working now.

**SOLUTION:**
Rebuilt the program after adding the following lines in the example's CMakeLists.txt

pico_enable_stdio_usb(picow_blink 1)
pico_enable_stdio_uart(picow_blink 0)

closed 2023-01-27 by d-adamson