raspberrypi/picotool #79: -f option missing for load command on Windows

ck2510, 2023-04-01:
As per README it is stated that the load command supports the "-f" option as of version 1.1.

When using the picotool version 1.1.1 on Windows this option still is absent. Also help does not show it and trying to use it results in error.
This is the help excerpt I get:
picotool load [-n] [-N] [-u] [-v] [-x] <filename> [-t <type>] [-o <offset>] [--bus <bus>] [--address <addr>]

Potzi123, 2024-01-24:
Did you fix the issue in the mean time?

GoldeggCoder, 2024-01-24:
I also face the same issue. I want to upload to my PicoW without always unplugging it and pressing the BOOTSEL button.

will-v-pi, 2024-08-09:
This has been added with picotool 2.0.0 - previously the `-f` option was only available for the reboot command on Windows, as it does not work with RP2040 on Windows for other commands. Now, it is available for all commands as it works on RP2350, but will throw an error if you use `-f` with RP2040 on Windows telling you to reboot with `picotool reboot -f -u` first

closed 2024-08-09 by will-v-pi