raspberrypi/picotool #81: The compiler or linker does not enter the information in the binary file.

tistructor, 2023-04-25:
I use the core: "https://github.com/earlephilhower/arduino-pico"
I followed the guide contained in the file: "getting-started-with-pico.pdf"
And the compilation takes place without errors.
Using picotool the information I entered is not visible, however.
I believe the compiler eliminates the two functions.
I looked at the file: "ctags_target_for_gcc_minus_e.cpp" which is generated at the end of the compilation and the two functions are not there.

This is the file :
```
/*
  Fade

  This example shows how to fade an LED on pin 9 using the analogWrite()
  function.

  The analogWrite() function uses PWM, so if you want to change the pin you're
  using, be sure to use another PWM capable pin. On most Arduino, the PWM pins
  are identified with a "~" sign, like ~3, ~5, ~6, ~9, ~10 and ~11.

  This example code is in the public domain.

  https://www.arduino.cc/en/Tutorial/BuiltInExamples/Fade
*/
#include "pico/binary_info.h"

int led = 25;           // the PWM pin the LED is attached to
int brightness = 0;    // how bright the LED is
int fadeAmount = 5;    // how many points to fade the LED by

// the setup routine runs once when you press reset:
void setup() {
  // declare pin 9 to be an output:
  pinMode(led, OUTPUT);
  
}

// the loop routine runs over and over again forever:
void loop() {
  bi_decl(bi_program_description("This is a test binary."));
  bi_decl(bi_1pin_with_name(LED_PIN, "On-board LED"));
  // set the brightness of pin 9:
  analogWrite(led, brightness);

  // change the brightness for next time through the loop:
  brightness = brightness + fadeAmount;

  // reverse the direction of the fading at the ends of the fade:
  if (brightness <= 0 || brightness >= 255) {
    fadeAmount = -fadeAmount;
  }
  // wait for 30 milliseconds to see the dimming effect
  delay(30);
```

I also put the functions in setup () but the functions are eliminated.

I have not understood what needs to be done to set up the compiler or linker to prevent functions from being eliminated.

I tried to remove from "platform.txt" --gc-sections from the parameters passed to the linker

```
compiler.ldflags={compiler.wrap} -Wl,--cref -Wl,--check-sections -Wl,--gc-sections,--unresolved-symbols=report-all -Wl,--warn-common
```
but in this way the compilation ends with errors.

lurch, 2023-04-25:
That might be a bug with https://github.com/earlephilhower/arduino-pico (which we don't officially support), rather than a bug with picotool?

tistructor, 2023-04-25:
Thanks,
I searched the internet and on your github for mid afternoon but didn't check on the core website.
 I only managed to find the example of the c code to be able to use it.

I'm sorry this picotool function is interesting.

We hope that in the future the arduino core will be modified to support this functionality.

Ciao

closed 2024-07-27 by will-v-pi