raspberrypi/picotool #82: add support for binaries behind boot loaders on device

kilograham, 2023-04-27:
If the boot loader itself contains binary info which points to the other binaries, we could support this

earlephilhower, 2023-04-27:
Sounds like a good plan!  As long as the `binary_info_header_redirect` (or whatever) is defined in such a way that it can be static even as its target (real) app changes, should be easy to support in OTA/2ndary bootloaders.  I suppose that's kind of a given since the `binary_info_header` of the real app is always placed immediately after its vector table, and normally the real app starting flash address is always constant too.

will-v-pi, 2024-11-20:
This is achieved for RP2350 in #173, which prints out info about the bootloader and the binaries in partitions