raspberrypi/picotool #93: msys2 package

thewh1teagle, 2024-01-04:
I created jew picotool package for msys2 and it's already in their packages

https://packages.msys2.org/package/mingw-w64-x86_64-picotool?repo=mingw64

You can change your instructions in readme so users can install it without compile

lurch, 2024-01-11:
I believe the recommended approach on Windows is to use https://github.com/raspberrypi/pico-setup-windows which you also "install without compile". But ping @nathan-contino anyway, just so that he's aware of this.

ndabas, 2024-01-11:
Indeed, pico-setup-windows includes a pre-built binary of picotool. However [the docs] still say that you need to build it yourself, so maybe we should update those?

thewh1teagle, 2024-01-11:
@ndabas @lurch 
I don't know if you tried yourself to use pico-setup for developing on windows, but it's very heavy and inconvenient.
I made this tutorial which takes 1-5 minutes for setup the whole SDK on windows.
https://github.com/thewh1teagle/pico-setup

ndabas, 2024-01-12:
> I don't know if you tried yourself to use pico-setup for developing on windows, but it's very heavy and inconvenient.

Did you mean pico-setup-windows?
1. Go to https://github.com/raspberrypi/pico-setup-windows.
2. Check the list of contributors.

I appreciate the effort you've put in, and appreciate the fact that some people might have a preference for installing tools into their preferred environment. However pico-setup-windows also installs in 2-3 minutes on most systems, installs the same tools that your script is installing, and I'm not sure what you mean by inconvenient. If you have some specific issues with that installer it would be great if you could let us know so we can fix those in future releases.