raspberrypi/picotool #94: Error saving all flash rom

mardini1974, 2024-02-12:
Hello, When I try to save all flash from the pico, it disconnects from BOOTSEL mode and I get an error.

[image]

I have to disconnect the pico and set it back to BOOTSEL mode, while I can save a program  (-p) without any problem.
i'm working in Windows WSL2 ubuntu, using usbipd to [connect USB device to ubuntu ](https://devblogs.microsoft.com/commandline/connecting-usb-devices-to-wsl/)

lurch, 2024-02-12:
This may not be relevant to your problem, but I think if your Pico is already in BOOTSEL mode, then there's no need to use the `-F` option of picotool?

mardini1974, 2024-02-12:
> This may not be relevant to your problem, but I think if your Pico is already in BOOTSEL mode, then there's no need to use the `-F` option of picotool?

I tried both with Force and without force, both are the same.

mardini1974, 2024-02-13:
Hi Kilograham, I fried the first Raspberry Pi PICO due to a faulty connection, but I did the test on a new PICOs and it worked correctly I could copy the RAM from one to the other, So the problem maybe in WSL or USBipd, The challenge in installing the binary on my work computer give some issues with IT security!, I do prefer building the application on WSL.

will-v-pi, 2024-07-29:
I've tried this on the latest WSL2 version (2.2.4) with the latest usbipd (4.2.0) and am unable to reproduce this issue - could you try updating to the latest versions and check if it still occurs?

will-v-pi, 2024-08-28:
Closing this for now, as I can't reproduce it - feel free to reopen if you're still having this issue

closed 2024-08-28 by will-v-pi