raspberrypi/picotool #95: Build on Windows with MinGW without WSL

micropicostack, 2024-04-03:
I tired to build the tool on Windows but the options documented in README.md did not work for me. 

I do not have a Visual Studio installation, so I wanted to use MinGW. After some time tinkering around I found that modifying the `FindLIBUSB.cmake` file and replacing the `HINTS $ENV{LIBUSB_ROOT}/VS2019/MS32/static` line with `HINTS $ENV{LIBUSB_ROOT}/MinGW64/static` and running `cmake -G "MinGW Makefiles" ..` followed by `mingw-make` forked like a charm, I got my exe.

Now to the question or idea - could we update the `FindLIBUSB.cmake` to consider 32bit and 64bit machines, Windows MSVC and MinGW lib inclusion based on generator input?

For example something like this ?

```cmake
if(CMAKE_GENERATOR MATCHES "MinGW")
    HINTS $ENV{LIBUSB_ROOT}/MinGW64/static
else
    HINTS $ENV{LIBUSB_ROOT}/VS2019/MS64/static
endif()
```

lurch, 2024-04-03:
> Now to the question or idea - could we update the `FindLIBUSB.cmake` to consider 32bit and 64bit machines, Windows MSVC and MinGW lib

Yes, we've done something similar before :slightly_smiling_face:  https://github.com/raspberrypi/picotool/commit/757f2aee3c2c988b4d0c9c15907e243350381d0a

PRs welcome!

ndabas, 2024-04-04:
If you're just looking to compile picotool on Windows without the overhead of Visual Studio or WSL, MSYS2 works fine with no changes, like [we've done for pico-setup-windows](https://github.com/raspberrypi/pico-setup-windows/blob/master/packages/picotool/build-picotool.sh).

micropicostack, 2024-04-07:
> > Now to the question or idea - could we update the `FindLIBUSB.cmake` to consider 32bit and 64bit machines, Windows MSVC and MinGW lib
> 
> Yes, we've done something similar before 🙂 [757f2ae](https://github.com/raspberrypi/picotool/commit/757f2aee3c2c988b4d0c9c15907e243350381d0a)
> 
> PRs welcome!

Okey, I might do a PR then, but I am not that familiar with Cmake. But to make the process as automatic as possible we have to 
- Determine if we are running MinGW/Cygwin or Visual Studio generator
- Determine if we are on a 32bit or a 64bit machine

As for Visual Studio folder I guess it will be fixed to 2019 (although it seems there are newer versions - 2022).

Does that sound like a plan?

micropicostack, 2024-04-07:
> If you're just looking to compile picotool on Windows without the overhead of Visual Studio or WSL, MSYS2 works fine with no changes, like [we've done for pico-setup-windows](https://github.com/raspberrypi/pico-setup-windows/blob/master/packages/picotool/build-picotool.sh).

I got it to work with MinGW64 binaries (v13.2.0), I had to let `cmake` know that I was using MinGW generator and I then I used `mingw32-make.exe` to build the executable. 

I made the issue as we could expand either on the `FindLIBUSB.cmake` and automate a little more or add more description to `README.md`

Sultan-papagani, 2024-06-20:
[image]

it worked thank you so much

micropicostack, 2024-07-15:
Created a PR https://github.com/raspberrypi/picotool/pull/102 
This should improve the tool build on Windows machines.

will-v-pi, 2024-08-09:
Thanks - #102 has been merged for picotool 2.0.0

closed 2024-08-09 by will-v-pi