raspberrypi/picotool #110: pico-examples inline download and build of picotool does not accurately detect libusb

DarwinJS, 2024-08-10:
When I update a project in pico-examples with usb support the build process is smart enough to download and build picotool as a part of the overall build.

However, the build reports: "libUSB is not found - no USB support will be built"

On the exact same dev machine, if I download this repository directly and build it - it builds fine with usb support (all dependencies are present).

It would be extremely handy if that integrated build accurately built with USB support because it would eliminate many tutorial steps to build picotool again separately.

I did try logging out and in again in case there was some type of path update issue after installing the OS dependencies (even though the full build directly from this project did not require that work).

Is there a configuration change I would need to make when building pico-examples to get this working?

DarwinJS, 2024-08-10:
Here is some potentially useful output:

```
sudo apt-cache search libusb
golang-github-google-gousb-dev - provides low-level interface for accessing USB devices (library)
golang-github-hanwen-usb-dev - CGO bindings for libusb
libhidapi-libusb0 - Multi-Platform library for communication with HID devices (libusb backend)
libhpmud-dev - HP Multi-Point Transport Driver (hpmud) development libraries
libhpmud0 - HP Multi-Point Transport Driver (hpmud) run-time libraries
libdevice-usb-perl - Perl module to access USB devices using libusb
gir1.2-fprint-2.0 - async fingerprint library of fprint project, gir bindings
libfprint-2-2 - async fingerprint library of fprint project, shared libraries
libfprint-2-dev - async fingerprint library of fprint project, development headers
libfprint-2-doc - async fingerprint library of fprint project, API documentation
libftdi1 - Library to control and program the FTDI USB controller
libftdi1-2 - C Library to control and program the FTDI USB controllers
libftdipp1-3 - C++ Library to control and program the FTDI USB controllers
python3-ftdi1 - Python module to control and program the FTDI USB controllers
libgusb-dev - GLib wrapper around libusb1 - development files
libgusb-doc - GLib wrapper around libusb1 - documentation
libgusb2 - GLib wrapper around libusb1
libusb-0.1-4 - userspace USB programming library
libusb-dev - userspace USB programming library development files
libusb-1.0-0 - userspace USB programming library
libusb-1.0-0-dev - userspace USB programming library development files
libusb-1.0-doc - documentation for userspace USB programming
libusb-java - Java wrapper for libusb
libusb-java-doc - Java wrapper for libusb (documentation)
libusb-java-lib - Java wrapper for libusb (native library)
libusb-libusb-perl - Perl interface to the libusb-1.0 API
libusb3380-0 - USB3380 abstraction layer for libusb
libusb3380-dev - USB3380 abstraction layer for libusb: development
libusbauth-configparser-dev - Development package of library for USB Firewall including flex/bison parser
libusbauth-configparser1 - Library for USB Firewall including flex/bison parser
libusbgx-dev - Development files for libusbgx
libusbgx-doc - Documentation files for libusbgx
libusbgx2 - C library encapsulating the kernel USB gadget-configfs
libusbmuxd-dev - USB multiplexor daemon for iPhone and iPod Touch devices - devel
libusbmuxd-tools - USB multiplexor daemon for iPhone and iPod Touch devices - tools
libusbmuxd6 - USB multiplexor daemon for iPhone and iPod Touch devices - library
libuvc-dev - cross-platform library for USB video devices - development files
libuvc-doc - cross-platform library for USB video devices - documentation
libuvc0 - cross-platform library for USB video devices
lm4flash - Command-line firmware flashing tool to communicate with the Stellaris Launchpad
libusb-ocaml - OCaml bindings to libusb-1.0 (runtime)
libusb-ocaml-dev - OCaml bindings to libusb-1.0
ponyprog - Serial device programmer
python3-libusb1 - transitional package
python3-usb1 - Python wrapper for libusb1 (Python 3)
spectools - Utilities for using the Wi-Spy USB spectrum analyzer hardware
libusbguard-dev - USB device authorization policy framework - development files
libusbguard1 - USB device authorization policy framework - shared library
libusbredirhost-dev - implementing the usb-host (*) side of a usbredir connection (development)
libusbredirhost1 - Implementing the usb-host (*) side of a usbredir connection (runtime)
libusbredirparser-dev - Parser for the usbredir protocol (development)
libusbredirparser1 - Parser for the usbredir protocol (runtime)
```

jackwilsdon, 2024-08-11:
This appears to be intentional: https://github.com/raspberrypi/pico-sdk/blob/efe2103f9b28458a1615ff096054479743ade236/tools/Findpicotool.cmake#L65

I assume you're probably not intended to directly use the version of picotool built by the SDK - it's just there for the uf2 / encryption tools.

jackwilsdon, 2024-08-12:
Provided libusb is available, I can't see any harm in re-enabling USB support.

will-v-pi, 2024-08-13:
Yes, this is intentional - when building as part of the SDK, picotool is compiled without libusb to reduce compilation time.

If you want a picotool you can use with USB then we’d recommend building it separately, and then see the README for the different ways you can get the SDK to use that version of picotool.

Alternatively, you can just delete that line - it shouldn’t cause any issues to build it with libusb support as part of the SDK

DarwinJS, 2024-08-13:
Thank you all!

I realized that part of my confusion is that when the support is forced off by the switch, the build warning is still "libUSB is not found - no USB support will be built"

I'm not a cmake expert or I would MR a separate message when the switch is used - something like "libUSB was disabled with -DPICOTOOL_NO_LIBUSB=1  - no USB support will be built"

will-v-pi, 2024-08-13:
I've just added that with the referenced commit 3783fb4

commit 3783fb42 (2024-08-13): Fix #110 - Improve message when PICOTOOL_NO_LIBUSB is set
closed 2024-08-28 by will-v-pi