raspberrypi/picotool #117: Installed picotool not found by cmake on RPi OS Bullseye

Memotech-Bill, 2024-08-22:
On RPi OS Bullseye 32-bit, running `sudo make install` in the **picotool/build** folder installs the following files:

````
Install the project...
-- Install configuration: "Release"
-- Installing: /usr/local/bin/picotool
-- Installing: /usr/local/lib/arm-linux-gnueabihf/cmake/picotool/picotoolTargets.cmake
-- Installing: /usr/local/lib/arm-linux-gnueabihf/cmake/picotool/picotoolTargets-release.cmake
-- Installing: /usr/local/lib/arm-linux-gnueabihf/cmake/picotool/picotoolConfig.cmake
-- Installing: /usr/local/lib/arm-linux-gnueabihf/cmake/picotool/picotoolConfigVersion.cmake
-- Installing: /usr/local/share/picotool/rp2350_otp_contents.json
-- Installing: /usr/local/share/picotool/xip_ram_perms.elf
````

However, when running **cmake** to build a project using the **pico-sdk**, **cmake** searches for the following files:

````
 find_package considered the following locations for the Config module:

    /home/pi/.local/picotoolConfig.cmake
    /home/pi/.local/picotool-config.cmake
    /usr/local/picotoolConfig.cmake
    /usr/local/picotool-config.cmake
    /usr/local/share/picotool/picotoolConfig.cmake
    /usr/local/share/picotool/picotool-config.cmake
    /usr/picotoolConfig.cmake
    /usr/picotool-config.cmake
    /picotoolConfig.cmake
    /picotool-config.cmake
    /usr/local/games/picotoolConfig.cmake
    /usr/local/games/picotool-config.cmake
    /usr/games/picotoolConfig.cmake
    /usr/games/picotool-config.cmake

  The file was not found.

Call Stack (most recent call first):
  /home/pi/pico/pico-sdk/src/cmake/on_device.cmake:33 (pico_init_picotool)
  /home/pi/pico/pico-sdk/src/rp2040/boot_stage2/CMakeLists.txt:57 (pico_add_dis_output)
  /home/pi/pico/pico-sdk/src/rp2040/boot_stage2/CMakeLists.txt:101 (pico_define_boot_stage2)

CMake Warning at /home/pi/pico/pico-sdk/tools/Findpicotool.cmake:28 (message):
  No installed picotool with version 2.0.0 found - building from source
````

These do not correspond to the location the **picotoolConfig.cmake** was copied to when **picotool** was installed. As a result **picotool** is downloaded and compiled again everytime a project is built.

For further discussion see https://forums.raspberrypi.com/viewtopic.php?t=375486

will-v-pi, 2024-08-23:
Looks like this was a bug/feature with GNUInstallDirs, fixed in CMake 3.23.1 (see https://gitlab.kitware.com/cmake/cmake/-/merge_requests/7127 and associated issues). I'll set the libdir manually in picotool, rather than using the GNUInstallDirs one, which should fix this

Memotech-Bill, 2024-08-23:
I am not sure that this will completely resolve the issue.

If I understand correctly, your revision will mean that `make install` will install **picotoolConfig.cmake** in **/usr/local/lib/cmake/picotool/** instead of **/usr/local/lib/arm-linux-gnueabihf/cmake/picotool/**.

However if you look at the list of locations (in my first post) that the Bullseye version of **SDK cmake** searched for **picotoolConfig.cmake**, it does not include **/usr/local/lib/cmake/picotool/**.

Memotech-Bill, 2024-08-23:
Tried it. The file was found, but it did not work. The complete output from **cmake** is in the [attached file](https://github.com/user-attachments/files/16732725/temp.txt), but the relevant sections are:

````
CMake Error at /usr/local/lib/cmake/picotool/picotoolTargets.cmake:71 (message):
  The imported target "picotool" references the file

     "/usr/bin/picotool"

  but this file does not exist.  Possible reasons include:

  * The file was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and contained

     "/usr/local/lib/cmake/picotool/picotoolTargets.cmake"

  but not all the files it references.

Call Stack (most recent call first):
  /usr/local/lib/cmake/picotool/picotoolConfig.cmake:2 (include)
  /home/pi/pico/pico-sdk/tools/CMakeLists.txt:123 (find_package)
  /home/pi/pico/pico-sdk/src/cmake/on_device.cmake:33 (pico_init_picotool)
  /home/pi/pico/pico-sdk/src/rp2040/boot_stage2/CMakeLists.txt:57 (pico_add_dis_output)
  /home/pi/pico/pico-sdk/src/rp2040/boot_stage2/CMakeLists.txt:101 (pico_define_boot_stage2)
````
=== Lines omitted ===

````
  find_package considered the following locations for the Config module:

    /home/pi/.local/picotoolConfig.cmake
    /home/pi/.local/picotool-config.cmake
    /usr/local/picotoolConfig.cmake
    /usr/local/picotool-config.cmake
    /usr/local/lib/cmake/picotool/picotoolConfig.cmake

  The file was found at

    /usr/local/lib/cmake/picotool/picotoolConfig.cmake

Call Stack (most recent call first):
  /home/pi/pico/pico-sdk/src/cmake/on_device.cmake:33 (pico_init_picotool)
  /home/pi/pico/pico-sdk/src/rp2040/boot_stage2/CMakeLists.txt:57 (pico_add_dis_output)
  /home/pi/pico/pico-sdk/src/rp2040/boot_stage2/CMakeLists.txt:101 (pico_define_boot_stage2)
````

If it is relevant, the project I was attempting to build was https://github.com/Memotech-Bill/MEMU, and the **cmake** command was:

````
cmake -DTARGET=Pico .. --debug-find-pkg=picotool >temp.txt 2>&1
````
Note that in order to run the test I just did:

````
cd /usr/local/lib/
sudo mv -v arm-linux-gnueabihf/cmake/ .
````

I have not downloaded the patched version of the **picotool** repository. I guess that your change also results in a change to the contents of either **picotoolConfig.cmake** or **picotoolTargets.cmake**.

will-v-pi, 2024-08-23:
Don’t move the files around, it breaks stuff - the links between the auto-generated CMake config files and the picotool binary are all relative, so moving them up one directory level means that the CMake files are no longer pointing to the correct location for picotool - hence them now expecting a binary in /usr/bin, when the binary is actually installed in /usr/local/bin.

To test this fix you’ll need to clone the develop branch of picotool, then run a normal `cmake..` `make` `sudo make install` build and install

Memotech-Bill, 2024-08-23:
My test was sufficient to prove that **/usr/local/lib/cmake/picotool/picotoolConfig.cmake** was found.

For now I am happy setting **picotool_DIR** environment variable.

I will wait for the next release of **picotool** and the SDK.

commit 6ad9c235 (2024-08-23): Use lib instead of CMAKE_INSTALL_LIBDIR

Fixes #117
closed 2024-08-28 by will-v-pi