raspberrypi/picotool #119: picotool segfaults when load pico2 but not with pico

tomdean1939, 2024-08-25:
To load a pico2, picotool needs sudo - yes, I copied 99-picotool.rules
> ls /etc/udev/rules.d/
99-SaleaeLogic.rules  99-picotool.rules
====== with pico2 ===================
## built with set(PICO_BOARD pico2)
> lsusb | grep Pico
Bus 009 Device 121: ID 2e8a:0009 Raspberry Pi Pico
> picotool load blink.uf2 -f
Segmentation fault (core dumped)
> sudo picotool load blink.uf2 -f
Tracking device serial number F5B22EFA0A7A108E for reboot
The device was asked to reboot into BOOTSEL mode so the command can be executed.
Family id 'rp2350-arm-s' can be downloaded in absolute space:
  00000000->02000000
Loading into Flash: [==============================]  100%

====== with pico ===================================
# built with set(PICO_BOARD pico)
> lsusb | grep Pico
Bus 009 Device 019: ID 2e8a:000a Raspberry Pi Pico
> picotool load blink.uf2 -f
Loading into Flash: [==============================]  100%
aorus> picotool load blink.uf2 -f
Tracking device serial number E6614103E73D5636 for reboot
The device was asked to reboot into BOOTSEL mode so the command can be executed.
Loading into Flash: [==============================]  100%
The device was asked to reboot back into application mode.

will-v-pi, 2024-08-26:
Could you provide the UF2 file you're using, and what platform you're running picotool on? Also, make sure your 99-picotool.rules file is up to date, as it was updated for RP2350, and you've rebooted since changing it - it should include the `0x0009` and `0x000F` PIDs, which are for the RP2350

tomdean1939, 2024-08-26:
No application should ever segfault. However, I did not reboot. I used 'sudo udevadm trigger' which should have reloaded
99-picotool.rules , which is the latest from github, after the RP2350 update.
Rebooting fixed the problem.
Again, there is the problem of segfault, which should never occur.
Sorry for the noise.

lurch, 2024-08-27:
@tomdean1939 Just to clarify: are you saying that since you rebooted, you've not been able to reproduce this segfault?

lurch, 2024-08-27:
Thanks for the confirmation. Closing this issue then.

closed 2024-08-27 by lurch