raspberrypi/picotool #120: My first Pico 2 SDK 2.0 issue - picotool

Some0neS0mewhere, 2024-08-25:
Hi All,

I've just run through the SDK Getting started guide using a Pico 2 and SDK 2.0 under visual studio code as mentioned in the getting started guide. This SDK works absolutely fine against the Pico 2 controller but whenever i try to use this SDK against the original Pico and Pico W and try to use the Run status bar button to upload to these units the below error is reported. This is after installing the Zadig app required for these legacy devices. The guide in the getting started doc is not descriptive enough on how to install Zadig as there is no dropdown option for picotool as described in this document. However, after installing this in the best way i can see the following error is reported.

[image]

I need advice on how to get past this issue when trying to use the legacy Pico and Pico W processors.

Techno11, 2024-08-26:
I am also experiencing this issue, exact same log.  Picos work when flashing via UF2, and with micropython, but not here.  Fresh install today, never had an older version on this PC.  Any advice is appreciated.

will-v-pi, 2024-08-26:
Looks like the getting started guide has incorrect instructions. The instructions that work for me for configuring Zadig are:
- Download Zadig from https://zadig.akeo.ie/ and run the executable
- Select "RP2 Boot (Interface 1)" from the device dropdown menu
- Select WinUSB from the driver dropdown menu (not libusb-win32)
- Click Install Driver
- Wait for Zadig to finish - this may take a few minutes

Let me know if those instructions work for you - I'll get the getting started guide updated. If you've already installed a different driver, you may need to click Options -> List All Devices, and then replace the existing driver with the WinUSB driver

Some0neS0mewhere, 2024-08-26:
Hi will-v-pi

I can confirm that changing or installing for the first time the WinUSB driver works under Windows 10 for the legacy Pico and Pico W devices.

closed 2024-08-29 by will-v-pi