raspberrypi/picotool #138: Please consider making fine grained commits

josch, 2024-09-09:
Example: 8a9af99ab10b20b1c6afb30cd9384e562a6647f9

>  83 changed files with 74,516 additions and 1,780 deletions. 

This makes it very hard to investigate how the tool changed, when each change was introduced and why, according to a git commit message. It also makes it impossible to bisect the history of picotool to find regressions.

Please consider actually working with git instead of making big code dumps. Otherwise, git is really pointless and you could as well make releases as tarballs without any history.

lurch, 2024-09-09:
That's because we (obviously) have to work on RP2350 in secret whilst it's in development, and so we work on picotool-for-RP2350 in an internal git repo. This internal git repo has **lots** of experimental commits / reversions etc. as the software gets developed in parallel with the hardware, and this would be _far_ too messy to release as a public commit-history (as well as possibly containing internal-only commits that we wouldn't ever want to be made public). So when the hardware gets released we simply take a "snapshot" of the current state of the internal git repo, and import that as a single (big) commit onto the public git repo. Yes, the internal-git-commit-history could _theoretically_ be rewritten into a suitable-for-public-consumption commit history, but there are so many internal commits that it would be a **gargantuan** job which would probably drive somebody insane :wink: (as well as delaying the release of RP2350, which we obviously wouldn't want to do).

The same thing is true (but even more so) for `pico-sdk` https://github.com/raspberrypi/pico-sdk/commit/efe2103f9b28458a1615ff096054479743ade236
> 933 changed files with 349,994 additions and 66,341 deletions.

Once the code is being developed out in the open on the public git repo, we _do_ then make fine-grained commits: https://github.com/raspberrypi/picotool/commits/develop/

josch, 2024-09-09:
Thank you for your elaborate reply!

Lets close this then. If somebody else asks, you have now something to link them to. :smile:

closed 2024-09-09 by josch