raspberrypi/picotool #142: please include AppStream metainfo XML

josch, 2024-09-09:
If you install this file into `/usr/share/metainfo/com.github.raspberrypi.picotool.metainfo.xml`:

```xml
<?xml version="1.0" encoding="UTF-8"?>
<component type="desktop">
  <id>com.github.raspberrypi.picotool</id>
  <metadata_license>MIT</metadata_license>
  <name>picotool</name>
  <summary>interact with RP2040 devices or binaries</summary>
  <description>
    <p>Tool for interacting with a RP2040 device in BOOTSEL mode, or
    with a RP2040 binary. It allows one to load programs onto the
    device or save programs from flash to a file. It allows one to
    verify device contents or reboot the device.</p>
  </description>
  <url type="homepage">https://github.com/raspberrypi/picotool/</url>
  <provides>
    <modalias>usb:v2E8Ap0003d*</modalias>
    <modalias>usb:v2E8Ap000Ad*</modalias>
  </provides>
</component>
```

Then tools like isenkram will be able to propose the installation of the picotool package on distributions that support this mechanism once a rp2040 is plugged into the user's system.

lurch, 2024-09-11:
Picotool is now useful for both RP2040 **and** RP2350? :wink:

josch, 2024-09-11:
> Picotool is now useful for both RP2040 **and** RP2350? 😉

Good point! What would be the correct modalias entry for the RP2350? You can find out by looking at the contents of the `modalias` file in the `/sys/devices` folder for the attached device.

lurch, 2024-09-11:
I don't have an RP2350 to hand right now, but those look suspiciously like USB PIDs, and you can find a full list of those at https://github.com/raspberrypi/usb-pid

josch, 2024-09-11:
Thank you! I adjusted the description to include rp2350 and added 0x000F as the rp2350 usb product id.

lurch, 2024-09-11:
If `0x000A` is included, you probably ought to include `0x0009` too? :thinking:

josch, 2024-09-12:
From the table I don't spot what the difference between the 0x000A and the 0x0009 rows is. The lower row has RP2040 appended to the description and the other does not. I do not have an RP2350 to see its usb product id.

lurch, 2024-09-12:
Using "git blame" I found https://github.com/raspberrypi/usb-pid/pull/18
Pre-emptively pinging @will-v-pi in case more information is needed?

will-v-pi, 2024-09-12:
`A` is for RP2040 only, as it behaves differently on Windows compared to RP2350 so needed a way to distinguish it (eg for `picotool … -f` commands). `9` is for RP2350, so both should probably be added to this file, else it won’t detect RP2350 devices using USB CDC

josch, 2024-09-12:
Thank you. I updated my original post. Please do with it what you like. I do not need any attribution for this. So if you want to ship this file, just somebody take the credit and commit it, thanks!