raspberrypi/picotool #145: Windows (non mingw): "the system cannot find the path specified: `picotool/build/rp2350_otp_contents.json`"

anandijain, 2024-09-19:
For context (and I'm not sure this is right) but I was hoping to avoid having to constantly unplug and plug my pico when trying to run/flash rust programs on my pico w. I don't know much about windows C compilation or any of this. 

But it seems like, from reading the doc of `picotool reboot` that this would make it easier to reboot the pico into a flashable state.

I am trying to build picotool. I am using the `vcpkg` LIBUSB and I should note I'm also compiling with 
```
C:\Users\anand\src\embedded\picotool\build>cmake -G "Visual Studio 17 2022" -A x64 -DPICO_SDK_PATH=C:\Users\anand\src\embedded\pico-sdk ..
```

Since I was getting a bunch of errors trying to just do the following (as the readme states): 
```
mkdir build
cd build
cmake -G "NMake Makefiles" ..
nmake
```

I got quite far into the build (I think) but I ran into the following failure which I am having trouble debugging 
```
  Configuring rp2350.rom.h
  Configuring xip_ram_perms_elf.h
  Building Custom Rule C:/Users/anand/src/embedded/picotool/CMakeLists.txt
  Generating C:/Users/anand/src/embedded/picotool/build/rp2350_otp_contents.json
  The system cannot find the path specified.
C:\Program Files\Microsoft Visual Studio\2022\Community\MSBuild\Microsoft\VC\v170\Microsoft.CppCommon.targets(254,5): e
rror MSB8066: Custom build for 'C:\Users\anand\src\embedded\picotool\build\CMakeFiles\6a64e9599d387ae51b1ed4000801235f\
rp2350_otp_contents.json.rule;C:\Users\anand\src\embedded\picotool\build\CMakeFiles\11b62aa6c3c80e9686a66e87350698ff\ge
nerate_otp_header.rule;C:\Users\anand\src\embedded\picotool\CMakeLists.txt' exited with code 3. [C:\Users\anand\src\emb
edded\picotool\build\generate_otp_header.vcxproj]
```

This seems related to the newly released rp2350, which I'm not using. Does anyone have any advice on how to resolve this issue? 

Thanks!

lurch, 2024-09-19:
I guess the easiest workaround is to just download the pre-compiled binary from https://github.com/raspberrypi/pico-sdk-tools

@will-v-pi As there have now been several similar requests for Windows binaries, perhaps it's worth adding that link somewhere near the top of https://github.com/raspberrypi/picotool/blob/master/README.md ?
CC @nathan-contino in case the getting-started docs need to make it clearer that pre-compiled binaries are actually available?

will-v-pi, 2024-09-19:
I've made a PR to add a note about the pre-compiled binaries to the readme. But it should be possible to compile on Windows - I've just checked and had success running:
```
cmake -G "Visual Studio 17 2022" -A x64 -DPICO_SDK_PATH=C:\path\to\pico-sdk -DLIBUSB_ROOT=C:\path\to\libusb-1.0.27\ ..
cmake --build .
```
So this may be an issue with the vcpkg libusb - could you try downloading libusb instead as mentioned in the instructions in the [readme](https://github.com/raspberrypi/picotool/#for-windows-without-mingw)?

Also make sure you either have `PICO_SDK_PATH` and `LIBUSB_ROOT` set as environment variables, or pass them to the cmake command

anandijain, 2024-09-19:
Hey guys, haven't gotten to the manual `libusb` stuff. But I just downloaded the binary and added it to my PATH and got it to work after an install of Zadig. 

Is my understanding correct that `picotool reboot --usb` will put my pico in BOOTSEL mode even if its running a program? 
Do I have to enable anything to make that work? It would be really convenient for me to not have to unplug the pico to reflash.

Otherwise I could just connect a digital pin to the RUN pin right?

will-v-pi, 2024-09-19:
You need to add the `-f` flag to your picotool command `picotool reboot -f -u`, and you need to be running compatible code - basically anything that uses stdio_usb, but see the [readme](https://github.com/raspberrypi/picotool?tab=readme-ov-file#forced-reboots) for more details