raspberrypi/picotool #148: picotool rejects valid .elf with `ERROR: 'path-to/hello-world-async.elf' failed validation - Unrecognized ABI`

kaspar030, 2024-09-20:
That .elf falls out of a regular rust embedded toolchain, linking was done with `rust-lld`.

I tracked that in the picotool sources to [this check](https://github.com/raspberrypi/picotool/blob/bf33c6ddd756c3961c7a13b000859b71e825ee74/elf/elf_file.cpp#L153-L155).
The failing binaries have the value set to `0x03` (meaning `UNIX - GNU` in `readelf -h` output).

Just hacking the field in the .elf header back to `0x00` makes the binary work fine.

While I have no clue why the linkers called through the Rust toolchain would set `0x03`, I believe the picotool check is overly restrictive, as the resulting binary works fine and is indeed using the correct ABI.

lurch, 2024-09-20:
Related to #111 ? Also pinging @thejpster for his formidable [Embedded Rust knowledge](https://www.raspberrypi.com/news/rust-on-rp2350/) :slightly_smiling_face:

kaspar030, 2024-09-20:
Yes that is exactly the issue. Thanks!

closed 2024-09-20 by kaspar030