raspberrypi/picotool #149: picotool not found when installed to the default location in /usr/local unless picotool_DIR is set

drankinatty, 2024-09-26:
This may simply be a documentation issue, but according to [README.md](https://github.com/raspberrypi/picotool/blob/master/README.md), setting the environment variable `picotool_DIR` is only necessary if installed to a custom path:

> Alternatively, you can install to a custom path via:
> 
> `cmake -DCMAKE_INSTALL_PREFIX=$MY_INSTALL_DIR -DPICOTOOL_FLAT_INSTALL=1 ..`
> 
> In order for the SDK to find picotool in this custom path, you will need to set the picotool_DIR variable ...

However, installing to the default location in `/usr/local` with `cmake --install .` results in `picotool` not being found and being repeatedly downloaded/built which slows building small projects down significantly over earlier versions.

This is detailed in the [Raspberry Pi Forum Thread - git pull/git submodule update --init -> warning: unable to rmdir](https://forums.raspberrypi.com/viewtopic.php?p=2254958#p2254974)

As noted in the thread, if it is necessary to set `picotool_DIR` regardless of how picotool is installed, then the `README.md` should be updated to make that clear. The way it is currently written implies setting the environment variable or including it in the `CMakeLists.txt` is only necessary if installed to a custom path location.

will-v-pi, 2024-09-26:
Looking at that forum thread, this is likely related to #117, which has been fixed in the `develop` branch of picotool. The issue is probably that the CMake files are being installed to `/usr/local/lib64/` but the `find_package` command in the pico-sdk is only searching `/usr/local/lib/`. Could you try building and installing from the `develop` branch of picotool and see if that helps?

will-v-pi, 2024-11-01:
Closing this as a duplicate of #117, which has been fixed in `develop`

closed 2024-11-01 by will-v-pi