raspberrypi/picotool #151: picotool silently segfaults if it can't open a device

usedbytes, 2024-10-02:
This is ~ a re-report of #119 , but I think the report there is unclear/misleading.

If picotool can't open the device (`ret  = libusb_open(device, dev_handle);` fails), then `handle` will end up as `0x0` in the device map, and (for example) `reboot_device` will segfault.

My Pico (RP2040) is reporting vid:pid `2e8a:000a` (Note the HW is actually a plain Pico, not a Badger2040, but that shouldn't matter)

```
$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 006: ID 2e8a:000a Pimoroni Badger2040
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
$ uname -a
Linux eddie 6.6.21-current-meson64 #1 SMP PREEMPT Wed Mar  6 14:48:45 UTC 2024 aarch64 aarch64 aarch64 GNU/Linux
$ ./picotool reboot -f
Segmentation fault
$ sudo ./picotool reboot -f
The device was asked to reboot into application mode.
```

At the very least I think the null handle should be checked/caught, rather than segfaulting.

lurch, 2024-10-03:
OffTopic: https://github.com/raspberrypi/usb-pid says 2e8a:000a is "Raspberry Pi Pico SDK CDC UART (RP2040)" so I've no idea why `lsusb` is listing it as "Pimoroni Badger2040" ??

usedbytes, 2024-10-03:
That’s just a string descriptor reported by my FW. `lsusb` doesn’t have a database or anything, it’s just printing what the device says in the Manufacturer and Product ID strings.

I originally wrote the code for a Badger 2040 and looks like I never changed the string.

will-v-pi, 2024-10-15:
I think I've found the cause of this segfault - could you test with that new branch and see if it fixes this for you? https://github.com/raspberrypi/picotool/tree/no-usb-segfault-fix

will-v-pi, 2024-10-29:
Merged into develop

PR #155 (2024-10-22): Fix segfault when unable to connect to stdio_usb device
This adds `dr_vidpid_stdio_usb_cant_connect` to the `picoboot_device_result` enum, to provide a more useful error when picotool is unable to connect to a device using stdio_usb. The current behaviour was a segfault in this situation, so this now fixes #151

This also builds on #154 to add the correct device to the error message when a stdio_usb device is discovered, rather than just printing RP-series.
commit c4550ad0 (2024-10-22): Fix segfault when unable to connect to stdio_usb device (#155)

Add dr_vidpid_stdio_usb_cant_connect when searching for devices, to detect USB devices that failed to open separately from ones that did

Fixes #151
closed 2024-10-29 by will-v-pi