raspberrypi/picotool #152: Flash RPI Pico Over GPIO Pins

mrmattrigby, 2024-10-07:
Is it possible to flash a Pico over the UART pins instead of over the serial USB connection? I have a project that will use several RPI Pico's, each connected over a mux to a controller with their UART0 TX and UART0 RX connected for message transferring. This eliminated managing USB port's on a server, but now updates to the design will require individually accessing each device.

Essentially, I was hoping for a "picotool -f" option with a UART select instead of over USB. Is this possible?

Thanks!

lurch, 2024-10-07:
AFAIK it's not possible to "flash" a Pico over UART (unless you've got some custom/3rd-party bootloader running on the Pico that provides that functionality), but the Pico 2 can "boot" over UART - see section 5.8. of https://datasheets.raspberrypi.com/rp2350/rp2350-datasheet.pdf

will-v-pi, 2024-10-15:
Yeah, this is not possible with picotool - you will need to use a custom bootloader. Picotool uses the USB Picoboot interface by the bootrom, not the USB Serial interface, so it can't simply connect to UART instead as the Pico Bootrom does not have a UART interface

closed 2024-10-15 by will-v-pi