raspberrypi/picotool #169: picotool uf2 convert cannot append arbitrary data

Athorus, 2024-11-08:
I’ve got 2 data files :
table.bin (283 bytes)
configuration.bin (64 bytes)

if I do
picotool uf2 convert -o 0x10001000 table.bin pico.uf2
picotool uf2 convert -o 0x10002000 configuration.bin pico.uf2
then I got :
a file pico.uf2 of 512 bytes long. It means table.bin has been removed from pico.uf2 (or pico.uf2 has been deleted before adding configuration.bin)

Thanks to picotool uf2 convert we should be able to add arbitrary data at arbitrary location.

I do not have this issue when I append binary codes compiled for the right location.

Athorus, 2024-11-08:
I was able to do the work with uf2conv.py tool
Would be great to do it with picotool uf2 convert

will-v-pi, 2024-11-08:
The `picotool uf2 convert` command does not append data to the uf2 file - it just creates a new UF2 file with the data you give it. Hence overwriting the `pico.uf2` file when you run the command a second time.

We could maybe add a `picotool uf2 combine` command to combine 2 UF2 files into one, but that wouldn't be a priority as you can just make 2 UF2 files then copy them one at a time.

Athorus, 2024-11-08:
You cannot concatenate uf2 files as they contain sequential numbers. Moreover, each time you copy an uf2 file to your pico, it reboots.
For my personnal case, it doesn’t matter as I’ve found another tool. I let you choose your priorities. ;)

lurch, 2024-11-11:
I guess this is a similar request to https://github.com/raspberrypi/pico-sdk/issues/1071 and https://github.com/raspberrypi/pico-sdk/issues/1300