raspberrypi/picotool #172: Document schema for all JSON file arguments

jakerr, 2024-11-12:
Please consider documenting what the schema is for all commands that take a JSON file.

While there are example JSON for some of them (partitions, white label, permissions), it would be ideal all JSON schema for each such argument was documented with a description of what each field in the JSON means, and what its possible values are.

will-v-pi, 2024-11-15:
I have added JSON schemas for all the JSON file inputs used in #176 - could you take a look at that and see if that's what you're looking for?

jakerr, 2024-11-16:
@will-v-pi Thank you so much for working on this, and so quickly at that! This is exactly the kind of documentation I had in mind. I think this really helps bridge the gap between the high-level overview in the datasheet's "Chapter 5. Bootrom" and what's actually required to implement some of those ideas using picotool.

I reviewed with my limited knowledge and it all looked good to me, but of course I don't know the ins and outs as I'm still learning.

Really appreciated!

PR #176 (2024-11-20): Add JSON schemas
Add JSON schemas for the JSON inputs used.

Currently the URLs point to the jsonschema branch - these should be changed to the develop branch once merged.

Fixes #172