raspberrypi/picotool #180: raspberry pi pico bootsel mode unable

heyitsmeyo, 2024-11-21:
Hello github people , i wanna put some attention on a problem that i m sure a lot of people here already faced when trying to enable bootsel mode on the rpi pico using picotool : 

No accessible RP2040 devices in BOOTSEL mode were found.

but:

Device at bus 3, address 12 appears to be a RP2040 device in BOOTSEL mode, but
    picotool was unable to connect.

I fortunately tried to make pico sdk project that can solve this issue : 

https://github.com/heyitsmeyo/rpi-pico-2040bootselmode 

use this command : 

sudo picotool load bootloader_reset.uf2 --bus <bus number> --address <address number> -f 

I hope it works ! 

Just a tip , whenever you work on pico sdk projects , make sure you include the stdio_usb.h library in your c file 

and on  your CMakeLists.txt : pico_enable_stdio_usb(<project name >  1)

will-v-pi, 2024-12-02:
Closing this, as I don't think that project is relevant to this issue - that project just loads a UF2 which reboots to bootsel mode, but if you're able to load that UF2 via picotool then you're able to reboot to bootsel mode simply using `sudo picotool reboot -f -u`, no need to load your `bootloader_reset.uf2` file.

Feel free to raise a PR if you think the documentation around [forced reboots](https://github.com/raspberrypi/picotool?tab=readme-ov-file#forced-reboots) can be improved

closed 2024-12-02 by will-v-pi

heyitsmeyo, 2024-12-02:
Thank you so much for the clarification ma friend , i didn't know that you can boot into bootsel mode like that , hahah i guess i made things difficult for me