raspberrypi/picotool #190: load command disappeared in picotool v2.1.0

A-Ibrahim, 2024-12-17:
load command disappeared
```picotool help load

ERROR: Unknown command: load

COMMANDS:
    info        Display information from the target device(s) or file.
                Without any arguments, this will display basic information for all connected RP-series devices in BOOTSEL mode
    config      Display or change program configuration settings from the target device(s) or file.
    encrypt     Encrypt the program.
    seal        Add final metadata to a binary, optionally including a hash and/or signature.
    link        Link multiple binaries into one block loop.
    otp         Commands related to the RP2350 OTP (One-Time-Programmable) Memory
    partition   Commands related to RP2350 Partition Tables
    uf2         Commands related to UF2 creation and status
    version     Display picotool version
    coprodis    Post-process coprocessor instructions in disassembly files.
    help        Show general help or help for a specific command

Use "picotool help <cmd>" for more info
```

will-v-pi, 2024-12-17:
You've compiled picotool without libUSB, so the USB commands (`load`, `save` etc) aren't available. Make sure you have `libusb-dev` installed then delete the build directory and recompile (see the readme for more details, including Windows instructions)

A-Ibrahim, 2024-12-18:
Thanks a lot this is solved my problem: 
`doas apt install libusb-1.0-0-dev -y`

closed 2024-12-18 by will-v-pi