raspberrypi/picotool #197: Unable to make Raspberry Pi Pico accessible without sudo (udev rule not working)

WilliamKarolDiCioccio, 2025-02-01:
I'm experiencing the same problem described in issue #32 and the posted fix seems not to be working anymore:

```txt
# make raspberry pi pico accessible without sudo.

SUBSYSTEM!="usb_device", ACTION!="add", GOTO="rpi2_end"
# Raspberry Pi Pico
ATTR{idVendor}=="2e8a", ATTRS{idProduct}=="0009", MODE="660", GROUP="plugdev"

LABEL="rpi2_end"
```

I've update the `ID_MODEL_ID` field according to the output of the `udevadm info /dev/ttyACM0` command.

I made sure that the newly added rules were applied after the `sudo udevadm control --reload-rules` command by unplugging and plugging the Pico 2 back in.

I'm running a fresh install of `Ubuntu 24.04.1 LTS`.

I've never worked in the Raspberry ecosystem and embedded in general so I'm probably the cause of the problem but as the available fix is not working for me I'm opening the issue in case other newcomers experience the same problem.

aescomputer, 2025-02-01:
When the Pico is connected normally on my system (after loading hello_usb), it reports this via `lsusb`
`Bus 008 Device 004: ID 2e8a:000a Raspberry Pi Pico`
and can be seen via `ls /dev | grep tty`
`ttyACM0`

But to load code from VS code, it has to be attached while holding the boot button, and when that is done, `lspci` shows:
`Bus 008 Device 002: ID 2e8a:0003 Raspberry Pi RP2 Boot`
so the udev rule needs to use this to enable loading code:
`ATTR{idVendor}=="2e8a", ATTRS{idProduct}=="0003", MODE="660", GROUP="plugdev"`

Also, mine did not work with reloading the rules and a reboot was needed.

WilliamKarolDiCioccio, 2025-02-02:
I've updated the rule to use product id "0003", after that I was able to flash the device but still unable to monitor the serial output. I opted to directly add myself to the `dialout` group:

`sudo usermod -a -G dialout $USER`

closed 2025-02-02 by WilliamKarolDiCioccio