raspberrypi/picotool #201: RP2350 device appears to be in BOOTSEL mode, but picotool was unable to connect.

makingpippop, 2025-02-11:
Hello,

I'm working on a new project with a Pico 2 and I want to create a completely separate environment on my machine. 
I'm using VSCode's DevContainers to create and run a Docker container with the complete toolchain. The problem is that when I try to run :
`sudo picotool info -a`
I get an error saying :
```
No accessible RP-series devices in BOOTSEL mode were found.

but:

RP2350 device at bus 1, address 2 appears to be in BOOTSEL mode, but picotool was unable to connect. Maybe try 'sudo' or check your permissions.
```

**What I've tried**
- I'm on Window, so in order to access the Pico 2 from my container I've followed these instructions to setup [usbipd-win](https://learn.microsoft.com/en-us/windows/wsl/connect-usb)
On my local machine I can see that the device is attached, and in the container I see the device listed when running : `lsusb`

- Following the advice in : _Getting started with Raspberry Pi Pico-series_ page 26, I've downloaded _Zadig_ and made sure that the driver for my device was _WinUSB_

- I added the _udev rule_ available in the picotool's repo. 

Any other ideas?
Thanks for taking the time!

The Dockerfile :
```
#the base image
FROM mcr.microsoft.com/devcontainers/rust:1-1-bullseye
#libraries location
ARG LOC_LIBRARIES="/usr/tmp"

#Install picotool dependencies (https://github.com/raspberrypi/picotool)
RUN apt-get update && apt-get install -y \
      udev \
      build-essential \
      pkg-config \
      libusb-1.0-0-dev \
      usbutils \
      cmake

#-download and install
WORKDIR "$LOC_LIBRARIES"
ENV PICO_SDK_PATH="${LOC_LIBRARIES}/pico-sdk"
RUN mkdir -p "$LOC_LIBRARIES" && \
      git clone --recurse-submodules https://github.com/raspberrypi/pico-sdk.git && \
      git clone --recurse-submodules https://github.com/raspberrypi/picotool.git && \
      cp ./udev/99-picotool.rules /etc/udev/rules.d/ && \
      cd picotool && \
      mkdir build && \
      cd build && \
      cmake .. && \
      make && \
      make install

#Install raspberry pi's build platform
RUN rustup target add thumbv8m.main-none-eabihf 

#the starting directory inside the container
WORKDIR /workspaces
```

lurch, 2025-02-11:
> I'm using VSCode's DevContainers to create and run a Docker container with the complete toolchain.

This isn't an officially-supported configuration. You're probably better off asking for help in the [forums](https://forums.raspberrypi.com/).

closed 2025-02-11 by lurch

makingpippop, 2025-02-11:
I did :  https://forums.raspberrypi.com/viewtopic.php?p=2293927

To continue my investigation, do you know any way I can get more information on what's causing the error?

lurch, 2025-02-12:
Personally I have no idea, as I don't use Windows. And I suspect that most Windows-users are just using https://github.com/raspberrypi/pico-vscode rather than using Docker?