raspberrypi/picotool #210: "ERROR: Found overlapping memory" on encrypted binary

richard9999999999, 2025-02-27:
If I run:
`picotool info -all ../build/hello_serial_enc.uf2`

I am getting following error:

```
File ../build/hello_serial_enc.uf2 family ID 'rp2350-arm-s':

ERROR: Found overlapping memory ranges 0x20000000->0x20003d5c and 0x20000000->20000014
```

I tried that with hello_serial_enc from pico examples and also with my own encrypted binary

richard9999999999, 2025-02-27:
GCC 10.3.1.

will-v-pi, 2025-02-28:
This is fixed in #207

richard9999999999, 2025-03-01:
I confirm that I don't see the problem anymore using picotool compiled from branch "picotool-encrypted-shares". Thank you.

richard9999999999, 2025-04-23:
I've reported few bugs related to encrypted binaries recently. Now most of them are solved in DEVELOP with exception of this one. Could this fix be also merged to DEVELOP so we will have a picotool version which works with enctypted binaries?

will-v-pi, 2025-04-23:
This fix is part of a larger PR which still needs more work, so you will need to wait for that PR to be merged, or cherry-pick the relevant commits from that PR. This only causes issues when reading info from encrypted binaries, so you can still compile and use encrypted binaries without this fix.

richard9999999999, 2025-04-23:
OK I see.. I guessed the problem is trivial to fix.

will-v-pi, 2025-06-02:
merged into develop

closed 2025-06-02 by will-v-pi