raspberrypi/picotool #211: Build Fails: rp2350_otp_contents.json missing

yavrohatgi, 2025-03-04:
I am trying to build picotool on Windows using Visual Studio 2019 and CMake, but the build fails due to a missing file: rp2350_otp_contents.json. The error I get is : [build_log.txt](https://github.com/user-attachments/files/19064330/build_log.txt)

I am using MSVC 19.29.30158 (Visual Studio 2019) with CMake Version: 3.31.6 and Python: 3.12.5.

lurch, 2025-03-04:
Looks like it's [this part](https://github.com/raspberrypi/picotool/blob/master/CMakeLists.txt#L146) that's failing. Have you got your `PICO_SDK_PATH` variable set correctly?

yavrohatgi, 2025-03-04:
Yes, I do have the PICO_SDK_PATH set correctly.

lurch, 2025-03-04:
...and does `${PICO_SDK_PATH}/src/rp2350/hardware_regs/include/hardware/regs/otp_data.h` exist?

tang-xiaolong-0219, 2025-03-20:
> I am trying to build picotool on Windows using Visual Studio 2019 and CMake, but the build fails due to a missing file: rp2350_otp_contents.json. The error I get is : [...]

Did you update the pico-sdk submodules?
I successfully compiled using ming64。

yavrohatgi, 2025-03-27:
Okay it worked, thanks [will-v-pi](https://github.com/will-v-pi)

closed 2025-03-27 by yavrohatgi