raspberrypi/picotool #215: Disable picotool functions on production build

codekrolik2, 2025-03-20:
Hello guys, I'm working on a custom open-source firmware for my RPi Pico, and it's supposed to ideally not be vulnerable to attacks related to replacing the original firmware with malicious versions, while connected via USB to the host computer (and it's usage suggests that is will be connected to USB most of the time).
While developing using VisualStudio Code and PlatformIO, I noticed that it's not even required to switch my device to any special mode for picotool to be able to force firmware update. Looks like if it's connected to a computer via USB, picotool can just arbitrarily re-flash its firmware at any time, and there's nothing that can be done about it.

I was trying to google how to prevent this from happening in release build and wasn't able to find anything.
Could you please help me understand if it's at all preventable?

codekrolik2, 2025-03-20:
I'm trying to build a security token repurposing this hardware: https://thumby.us/ so making sure the firmware stays immutable is kind of important

lurch, 2025-03-20:
> Looks like if it's connected to a computer via USB, picotool can just arbitrarily re-flash its firmware at any time

That's probably using the `PICO_STDIO_USB_ENABLE_RESET_VIA_VENDOR_INTERFACE` option which you can easily disable - see https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-c-sdk.pdf

> making sure the firmware stays immutable is kind of important

In that case you'll want to use a Pico 2 instead of a Pico 1, and read chapter 10 of https://datasheets.raspberrypi.com/rp2350/rp2350-datasheet.pdf

codekrolik2, 2025-03-20:
@lurch I've added the following to one of my h-files, and it didn't have any effect
```
#define PICO_STDIO_USB_ENABLE_RESET_VIA_VENDOR_INTERFACE 0
```
It this how I disable it?

codekrolik2, 2025-03-20:
```
[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git#196d31bbafaf60b84751b1a415d8dca2365debdf
board = rpipico
framework = arduino
board_build.filesystem_size = 0m
board_build.f_cpu = 125000000L
lib_deps = 
	https://github.com/TinyCircuits/TinyCircuits-Thumby-Lib/archive/refs/heads/master.zip
	https://github.com/TinyCircuits/TinyCircuits-GraphicsBuffer-Lib/archive/refs/heads/master.zip
	agdl/Base64@^1.0.0
platform_packages = 
	framework-arduinopico @ https://github.com/earlephilhower/arduino-pico.git#2.2.1
build_flags = -DPICO_STDIO_USB_ENABLE_RESET_VIA_VENDOR_INTERFACE=0 -DPICO_STDIO_USB_ENABLE_RESET_VIA_BAUD_RATE=0
```

lurch, 2025-03-20:
Yes - as long as your own header file where you `#define PICO_STDIO_USB_ENABLE_RESET_VIA_VENDOR_INTERFACE` gets included _before_ any of the pico-sdk header files, especially https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/pico_stdio_usb/include/pico/stdio_usb.h#L93
And of course you have to have the firmware with that setting disabled actually running on your Pico before it has any effect 😉 

I'm afraid that PlatformIO and Arduino are not platforms that we officially support ourselves - you'll have to ask for help on their forums / issue-trackers.

codekrolik2, 2025-03-20:
I tried this,

```
    #ifdef PICO_STDIO_USB_ENABLE_RESET_VIA_VENDOR_INTERFACE
      Serial.printf("PICO_STDIO_USB_ENABLE_RESET_VIA_VENDOR_INTERFACE is defined as: %d\n", PICO_STDIO_USB_ENABLE_RESET_VIA_VENDOR_INTERFACE);
    #else
      Serial.printf("PICO_STDIO_USB_ENABLE_RESET_VIA_VENDOR_INTERFACE is not defined.\n");
    #endif

    #ifdef PICO_STDIO_USB_ENABLE_RESET_VIA_BAUD_RATE
      Serial.printf("PICO_STDIO_USB_ENABLE_RESET_VIA_BAUD_RATE is defined as: %d\n", PICO_STDIO_USB_ENABLE_RESET_VIA_BAUD_RATE);
    #else
      Serial.printf("PICO_STDIO_USB_ENABLE_RESET_VIA_BAUD_RATE is not defined.\n");
    #endif
```
and the output was:

```
PICO_STDIO_USB_ENABLE_RESET_VIA_VENDOR_INTERFACE is defined as: 0                      
PICO_STDIO_USB_ENABLE_RESET_VIA_BAUD_RATE is defined as: 0
```

Looks like it's set to 0, just doesn't have any effect, picotool still works.

lurch, 2025-03-20:
> Looks like it's set to 0, just doesn't have any effect, picotool still works.

That seems to suggest that you're defining the value _after_ https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/pico_stdio_usb/reset_interface.c#L174 has already been run?

codekrolik2, 2025-03-20:
I removed that from my .h file actually, and use this:
```
build_flags = -DPICO_STDIO_USB_ENABLE_RESET_VIA_VENDOR_INTERFACE=0 -DPICO_STDIO_USB_ENABLE_RESET_VIA_BAUD_RATE=0 -DPICO_STDIO_USB_RESET_INTERFACE_SUPPORT_RESET_TO_BOOTSEL=0 -DPICO_STDIO_USB_RESET_INTERFACE_SUPPORT_RESET_TO_FLASH_BOOT=0

```

codekrolik2, 2025-03-20:
Output without build options:
```
PICO_STDIO_USB_ENABLE_RESET_VIA_VENDOR_INTERFACE is not defined.
PICO_STDIO_USB_ENABLE_RESET_VIA_BAUD_RATE is not defined.
PICO_STDIO_USB_RESET_INTERFACE_SUPPORT_RESET_TO_BOOTSEL is not defined.
PICO_STDIO_USB_RESET_INTERFACE_SUPPORT_RESET_TO_FLASH_BOOT is not defined.
```

Output with build options:
```
PICO_STDIO_USB_ENABLE_RESET_VIA_VENDOR_INTERFACE is defined as: 0                      
PICO_STDIO_USB_ENABLE_RESET_VIA_BAUD_RATE is defined as: 0                             
PICO_STDIO_USB_RESET_INTERFACE_SUPPORT_RESET_TO_BOOTSEL is defined as: 0               
PICO_STDIO_USB_RESET_INTERFACE_SUPPORT_RESET_TO_FLASH_BOOT is defined as: 0 
```

lurch, 2025-03-20:
As already mentioned, I don't know how the arduino or platformio build_flags work, so you'll have to ask them for help.

codekrolik2, 2025-03-20:
Ok, thanks again.
Please consider disabling such dangerous functions by default, to enable serious applications.

will-v-pi, 2025-03-20:
These options will not be disabled by default, as they are very useful for uploading code to a running device, and can be easily disabled for production by adding the following to your CMakeLists.txt when using the supported CMake build system (I'm not sure what PlatformIO uses, but there should be a way to add compile definitions like this)
```
target_compile_definitions(<target> PRIVATE
        PICO_STDIO_USB_ENABLE_RESET_VIA_VENDOR_INTERFACE=0
        PICO_STDIO_USB_ENABLE_RESET_VIA_BAUD_RATE=0
        )
```
These compile options are all documented in section 5.1 of https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-c-sdk.pdf

closed 2025-03-20 by will-v-pi

codekrolik2, 2025-03-20:
or easily forgotten

will-v-pi, 2025-03-20:
We would have many more people complaining if we disabled this very useful feature by default…

Also, even with those features disabled, it’s trivially easy to put the RP2040 in bootsel mode by holding BOOTSEL while plugging it in, and then you can upload firmware the same way picotool does.

If you want something secure that won’t run malicious firmware, get an RP2350 and enable secure boot.

codekrolik2, 2025-03-21:
I clearly understand that I have options of throwing all my Thumbys away and buying something else, trying to avoid RP2040-based stuff as much as possible. That fact certainly is not lost on me.

I guess my point is that some unsafe defaults from the sdk can get lost in stack, like PlatformIO level or smth. and getting an advice to fix those problems via unreachable compilatoin flags would be akin to a Ubuntu desktop user getting an advice to fix their PDF viewer by rebuilding their Linux kernel with different compilation flags.
So the reason that those defaults don't get fixed is because they are defaults provided by you, and since everybody trusts you and have no reason to doubt the sensibility of said defaults. I can imagine that when they support like over 9000 boards I bet they wont read every flag and default value in your manual.
But sounds like you believe that everybody should do just that. Well, the opinions can be different I guess.