raspberrypi/picotool #223: Build error starting with `2.1.0` on Windows (MingGW32 & Git Bash)

recursivenomad, 2025-04-06:
Picotool compiles and executes successfully in my build environment with version `2.0.0` (same SDK version).  Updating to `2.1.0` or `2.1.1` for both results in the following error when executing `make`:

```
C:/mingw32/bin/../lib/gcc/i686-w64-mingw32/14.2.0/../../../../i686-w64-mingw32/bin/ld.exe: C:/VSARM/sdk/pico/libusb/MinGW32/static/libusb-1.0.a(core.o):core.c:(.text+0x121): undefined reference to `_imp___iob'
```

I am curious if this could be related to PR #126?  Seems like that was a significant change to how libusb is referenced, and this error relates to what [Stack Overflow would suggest](https://stackoverflow.com/questions/66314384/undefined-reference-to-imp-in-minwg) might be static vs. dynamic linking.

For reference, my build environment is an updated version of what I outline in [this repo](https://github.com/recursivenomad/capacitance-measurement-module/blob/main/BUILDING.md).  Note that the versions linked there are outdated, but the process remains the same.

TLDR: I use Git Bash as my terminal, with a manually installed MinGW32.  I have a hunch that linking against the static `libusb-1.0.a` has broken my build, but I am unsure of the appropriate solution.

Let me know if I can provide any further information :)

recursivenomad, 2025-04-07:
Update: I have confirmed that Picotool builds successfully in my environment if you comment out the single-line change from #126.

recursivenomad, 2025-04-08:
> Could you confirm the mingw and libusb versions you are using?

`gcc --version` returns `gcc.exe (i686-mcf-dwarf-rev2, Built by MinGW-Builds project) 14.2.0`

libusb is extracted from [libusb-1.0.28.7z](https://github.com/libusb/libusb/releases/tag/v1.0.28)

*(I also experienced this issue when I was using mingw 13.2 before updating during troubleshooting)*

> That change was made to create statically linked binaries

So I'm unfamiliar with the nuances of static vs dynamic linking, but after re-reading your linked PR I can confirm that removing OpenOCD from my PATH results in picotool not playing nice (OpenOCD has libusb-1.0.dll in it).  Good to know... Agreed that maintaining static linking would be preferred.

> could add a guard around it if there is one to distinguish between the configurations which work with static linking and the ones that don't

I would certainly prefer discovering a solution which enables my build environment to function as expected for static linking, but a switch on this single line would be a sufficient enough workaround for my needs.

will-v-pi, 2025-04-08:
Looks like this might be similar to https://github.com/libusb/libusb/issues/574 - the solution there was to recompile libusb with your environment, to ensure it has the correct static variables. The libusb Windows binaries are provided as-is, with the caveats that they may not work with your particular environment, so I don't think there's anything we can do from the picotool side to fix the compilation.

We could add a CMake flag to disable that string replacement (eg `USE_LIBUSB_DLLS`) - if that's working for you then feel free to raise a PR to add that.

I would recommend compiling in WSL instead and then passing usb devices through to WSL using usbipd. Alternatively to compile it on Windows, you could use msys2 like we do in pico-sdk-tools, or use the Microsoft Visual Studio compiler.

recursivenomad, 2025-04-10:
I [was able](https://github.com/libusb/libusb/discussions/1629#discussioncomment-12769408) to get libusb to build by editing the following line from its `configure` file:

```diff
# Always use our own libtool.
- LIBTOOL='$(SHELL) $(top_builddir)/libtool'
+ LIBTOOL='"$(SHELL)" $(top_builddir)/libtool'
```

This allowed me to successfully build a statically linked picotool 🎉  Another solution I haven't done yet but should also work is to reinstall Git for Windows to a path without spaces.

So I personally no longer have a need for this change, but since others may prefer to compile with the as-is binaries (and I believe that is what is recommended in the README for Windows), this could still be a valid issue; I defer its status to you 👍

PR #228 (2025-06-18): Allow DLL libusb libraries with variable
Add `PICOTOOL_LIBUSB_ALLOW_DLL` variable to switch back to the old behaviour and use DLL libusb, which was used prior to #126

Fixes #223