raspberrypi/picotool #224: Cannot plug gap greater than alignment

richard9999999999, 2025-04-09:
With the current development branch of picotool I am getting following  error:

`[build] ERROR: Cannot plug gap greater than alignment - gap 231904, alignment 0`

I am using "wifi partition" enabled branch "will-v-pi:pico2-w-pts" (not sure if related). A short time ago it worked OK so the problem has been introduced by some of last commits most probably..

will-v-pi, 2025-04-09:
I'm unable to reproduce this issue when compiling the OTA updater with that SDK branch - could you provide your ELF file, or the output from `readelf -e`, to see what the issue is?

Note that a bug relating to this was fixed recently in 8f349d4840325b020129af5de566bf8bcafbf27c

richard9999999999, 2025-04-09:
```
d:\EFIS\build>readelf -e efis.elf
ELF Header:
  Magic:   7f 45 4c 46 01 01 01 00 00 00 00 00 00 00 00 00
  Class:                             ELF32
  Data:                              2's complement, little endian
  Version:                           1 (current)
  OS/ABI:                            UNIX - System V
  ABI Version:                       0
  Type:                              EXEC (Executable file)
  Machine:                           ARM
  Version:                           0x1
  Entry point address:               0x20000179
  Start of program headers:          52 (bytes into file)
  Start of section headers:          3738601 (bytes into file)
  Flags:                             0x5000200, Version5 EABI, soft-float ABI
  Size of this header:               52 (bytes)
  Size of program headers:           32 (bytes)
  Number of program headers:         5
  Size of section headers:           40 (bytes)
  Number of section headers:         30
  Section header string table index: 27

Section Headers:
  [Nr] Name              Type            Addr     Off    Size   ES Flg Lk Inf Al
  [ 0]                   NULL            00000000 000000 000000 00      0   0  0
  [ 1] .text             PROGBITS        20000000 001000 030b40 00  AX  0   0  8
  [ 2] .rodata           PROGBITS        20030b40 031b40 006bf8 00   A  0   0  8
  [ 3] .ARM.exidx        ARM_EXIDX       20037738 038738 000008 00  AL  1   0  4
  [ 4] .binary_info      PROGBITS        20037740 038740 000018 00   A  0   0  4
  [ 5] .data             PROGBITS        20037758 038758 000b80 00  WA  0   0  8
  [ 6] .tdata            PROGBITS        200382d8 0392d8 000000 00 WAT  0   0  1
  [ 7] .uninitialized_da PROGBITS        200382d8 0392d8 000000 00   W  0   0  1
  [ 8] .tbss             NOBITS          200382d8 000000 000000 00 WAT  0   0  1
  [ 9] .bss              NOBITS          200382d8 0392d8 015b3c 00  WA  0   0  8
  [10] .heap             PROGBITS        2004de14 0392d8 000800 00      0   0  4
  [11] .scratch_x        PROGBITS        20080000 039ad8 000000 00   W  0   0  1
  [12] .scratch_y        PROGBITS        20081000 039ad8 000000 00   W  0   0  1
  [13] .stack1_dummy     NOBITS          20080000 03a000 000180 00  WA  0   0  4
  [14] .stack_dummy      PROGBITS        20081000 039ae0 000800 00      0   0 32
  [15] .debug_info       PROGBITS        00000000 03a2e0 14e509 00      0   0  1
  [16] .debug_abbrev     PROGBITS        00000000 1887e9 01adcc 00      0   0  1
  [17] .debug_loc        PROGBITS        00000000 1a35b5 0c73e2 00      0   0  1
  [18] .debug_aranges    PROGBITS        00000000 26a998 003d08 00      0   0  8
  [19] .debug_ranges     PROGBITS        00000000 26e6a0 039210 00      0   0  8
  [20] .debug_line       PROGBITS        00000000 2a78b0 08b2a9 00      0   0  1
  [21] .debug_str        PROGBITS        00000000 332b59 033a40 01  MS  0   0  1
  [22] .comment          PROGBITS        00000000 366599 000049 01  MS  0   0  1
  [23] .ARM.attributes   ARM_ATTRIBUTES  00000000 3665e2 000036 00      0   0  1
  [24] .debug_frame      PROGBITS        00000000 366618 00ebac 00      0   0  4
  [25] .symtab           SYMTAB          00000000 3751c4 012040 10     26 3476  4
  [26] .strtab           STRTAB          00000000 387204 009738 00      0   0  1
  [27] .shstrtab         STRTAB          00000000 39093c 00012d 00      0   0  1
  [28] .pt               PROGBITS        200382d8 390a69 00008c 00   A  0   0  0
  [29] .sigx             PROGBITS        20038364 390af5 0000f4 00   A  0   0  0
Key to Flags:
  W (write), A (alloc), X (execute), M (merge), S (strings), I (info),
  L (link order), O (extra OS processing required), G (group), T (TLS),
  C (compressed), x (unknown), o (OS specific), E (exclude),
  y (purecode), p (processor specific)

Program Headers:
  Type           Offset   VirtAddr   PhysAddr   FileSiz MemSiz  Flg Align
  EXIDX          0x038738 0x20037738 0x20037738 0x00008 0x00008 R   0x4
  LOAD           0x001000 0x20000000 0x20000000 0x382d8 0x4de14 RWE 0x1000
  LOAD           0x000000 0x20080000 0x20080000 0x00000 0x00180 RW  0x1000
  LOAD           0x390a69 0x200382d8 0x200382d8 0x0008c 0x0008c R   0x2
  LOAD           0x390af5 0x20038364 0x20038364 0x000f4 0x000f4 R   0x2

 Section to Segment mapping:
  Segment Sections...
   00     .ARM.exidx
   01     .text .rodata .ARM.exidx .binary_info .data .bss
   02     .stack1_dummy
   03     .pt
   04     .sigx
```

will-v-pi, 2025-04-09:
Yes, this is the bug that was fixed in https://github.com/raspberrypi/picotool/commit/8f349d4840325b020129af5de566bf8bcafbf27c, so you were probably using an outdated picotool - check you're installing it correctly for the SDK to find it, see the [README](https://github.com/raspberrypi/picotool?tab=readme-ov-file#usage-by-the-raspberry-pi-pico-sdk) for more details

richard9999999999, 2025-04-09:
Aha, I found what makes the difference: I've had pico_encrypt_binary commented out (because I debugged something before). Now after enabling the encryption, the problem is back.. Sorry I needed to mention you that I am using encrypted binary..

richard9999999999, 2025-04-09:
So please check again with encrypted binary.

will-v-pi, 2025-04-09:
Could you post the readelf output from the problem binary, and preferably also attach the problem binary - the readelf above should only throw that error before that fix was added

richard9999999999, 2025-04-09:
```
[build] [ 10%] Linking CXX executable efis.elf
[build] ERROR: Cannot plug gap greater than alignment - gap 230496, alignment 0
[build] mingw32-make.exe[3]: *** [CMakeFiles\efis.dir\build.make:2725: efis.elf] Error -3
[build] mingw32-make.exe[3]: *** Deleting file 'efis.elf'
```
It deletes the elf file because unsuccessful build. Where can I comment that out?

will-v-pi, 2025-04-09:
You can use the Ninja generator instead of Makefiles (`cmake -GNinja`), and it will leave the ELF file behind even on a failed build

richard9999999999, 2025-04-09:
`ELF Header:
  Magic:   7f 45 4c 46 01 01 01 00 00 00 00 00 00 00 00 00
  Class:                             ELF32
  Data:                              2's complement, little endian
  Version:                           1 (current)
  OS/ABI:                            UNIX - System V
  ABI Version:                       0
  Type:                              EXEC (Executable file)
  Machine:                           ARM
  Version:                           0x1
  Entry point address:               0x20000179
  Start of program headers:          52 (bytes into file)
  Start of section headers:          3738959 (bytes into file)
  Flags:                             0x5000200, Version5 EABI, soft-float ABI
  Size of this header:               52 (bytes)
  Size of program headers:           32 (bytes)
  Number of program headers:         9
  Size of section headers:           40 (bytes)
  Number of section headers:         34
  Section header string table index: 27

Section Headers:
  [Nr] Name              Type            Addr     Off    Size   ES Flg Lk Inf Al
  [ 0]                   NULL            00000000 000000 000000 00      0   0  0
  [ 1] .text             PROGBITS        20000000 001000 030b40 00  AX  0   0  8
  [ 2] .rodata           PROGBITS        20030b40 031b40 006bf8 00   A  0   0  8
  [ 3] .ARM.exidx        ARM_EXIDX       20037738 038738 000008 00  AL  1   0  4
  [ 4] .binary_info      PROGBITS        20037740 038740 000018 00   A  0   0  4
  [ 5] .data             PROGBITS        20037758 038758 000b80 00  WA  0   0  8
  [ 6] .tdata            PROGBITS        200382d8 0392d8 000000 00 WAT  0   0  1
  [ 7] .uninitialized_da PROGBITS        200382d8 0392d8 000000 00   W  0   0  1
  [ 8] .tbss             NOBITS          200382d8 000000 000000 00 WAT  0   0  1
  [ 9] .bss              NOBITS          200382d8 0392d8 015b3c 00  WA  0   0  8
  [10] .heap             PROGBITS        2004de14 0392d8 000800 00      0   0  4
  [11] .scratch_x        PROGBITS        20080000 039ad8 000000 00   W  0   0  1
  [12] .scratch_y        PROGBITS        20081000 039ad8 000000 00   W  0   0  1
  [13] .stack1_dummy     NOBITS          20080000 03a000 000180 00  WA  0   0  4
  [14] .stack_dummy      PROGBITS        20081000 039ae0 000800 00      0   0 32
  [15] .debug_info       PROGBITS        00000000 03a2e0 14e509 00      0   0  1
  [16] .debug_abbrev     PROGBITS        00000000 1887e9 01adcc 00      0   0  1
  [17] .debug_loc        PROGBITS        00000000 1a35b5 0c73e2 00      0   0  1
  [18] .debug_aranges    PROGBITS        00000000 26a998 003d08 00      0   0  8
  [19] .debug_ranges     PROGBITS        00000000 26e6a0 039210 00      0   0  8
  [20] .debug_line       PROGBITS        00000000 2a78b0 08b2a9 00      0   0  1
  [21] .debug_str        PROGBITS        00000000 332b59 033a40 01  MS  0   0  1
  [22] .comment          PROGBITS        00000000 366599 000049 01  MS  0   0  1
  [23] .ARM.attributes   ARM_ATTRIBUTES  00000000 3665e2 000036 00      0   0  1
  [24] .debug_frame      PROGBITS        00000000 366618 00ebac 00      0   0  4
  [25] .symtab           SYMTAB          00000000 3751c4 012040 10     26 3476  4
  [26] .strtab           STRTAB          00000000 387204 009738 00      0   0  1
  [27] .shstrtab         STRTAB          00000000 39093c 00014f 00      0   0  1
  [28] .pt               PROGBITS        200382d8 390a8b 00008c 00   A  0   0  0
  [29] .sigx             PROGBITS        20038364 390b17 0000f4 00   A  0   0  0
  [30] .enc_pad          PROGBITS        20038458 390c0b 000008 00   A  0   0  0
  [31] .enc_link         PROGBITS        20000000 390c13 000014 00   A  0   0  0
  [32] .enc_iv           PROGBITS        20000014 390c27 000010 00   A  0   0  0
  [33] .sigx0            PROGBITS        20038484 390c37 000118 00   A  0   0  0
Key to Flags:
  W (write), A (alloc), X (execute), M (merge), S (strings), I (info),
  L (link order), O (extra OS processing required), G (group), T (TLS),
  C (compressed), x (unknown), o (OS specific), E (exclude),
  y (purecode), p (processor specific)

Program Headers:
  Type           Offset   VirtAddr   PhysAddr   FileSiz MemSiz  Flg Align
  EXIDX          0x038738 0x20037738 0x2003775c 0x00008 0x00008 R   0x4
  LOAD           0x001000 0x20000000 0x20000024 0x382d8 0x4de14 RWE 0x1000
  LOAD           0x000000 0x20080000 0x20080024 0x00000 0x00180 RW  0x1000
  LOAD           0x390a8b 0x200382d8 0x200382fc 0x0008c 0x0008c R   0x2
  LOAD           0x390b17 0x20038364 0x20038388 0x000f4 0x000f4 R   0x2
  LOAD           0x390c0b 0x20038458 0x2003847c 0x00008 0x00008 R   0x2
  LOAD           0x390c13 0x20000000 0x20000000 0x00014 0x00014 R   0x2
  LOAD           0x390c27 0x20000014 0x20000014 0x00010 0x00010 R   0x2
  LOAD           0x390c37 0x20038484 0x20038484 0x00118 0x00118 R   0x2

 Section to Segment mapping:
  Segment Sections...
   00     .ARM.exidx
   01     .text .rodata .ARM.exidx .binary_info .data .bss
   02     .stack1_dummy
   03     .pt
   04     .sigx
   05     .enc_pad
   06     .enc_link
   07     .enc_iv
   08     .sigx0`

will-v-pi, 2025-04-09:
I've implemented a fix on the `seg-fix` branch https://github.com/raspberrypi/picotool/commit/568e566a8414df43fe0fc3247a2a376071321c7c - the issue was due to the multiple matching virtual addresses in encrypted binaries. Let me know if that fixes it for you

richard9999999999, 2025-04-09:
I guess it you take example binary from "encrypted bootloader" sample, you will be able to reproduce the problem as well..

richard9999999999, 2025-04-09:
>  Let me know if that fixes it for you
OK, thank you

richard9999999999, 2025-04-09:
I can confirm, the problem seems to be gone with picotool built from `seg-fix` branch.

richard9999999999, 2025-04-11:
So if possible, please merge the change to develop so I will have the pitoctool usable..

will-v-pi, 2025-04-15:
See #227

will-v-pi, 2025-04-22:
merged into develop

commit c0d69419 (2025-04-15): Fix plugging alignment holes with encrypted binaries

Use new `segment_from_section` function, as virtual addresses are not unique per segment in encrypted binaries

Fixes #224
PR #227 (2025-04-22): Fix plugging alignment holes for encrypted binaries
The initial implementation missed the fact that for encrypted binaries, the virtual address 0x20000000 was in both the `.text` segment and the `.enc_link` segments, so they would show as in the same segment when they are not

Fixed by adding new `segment_from_section` function to definitively locate which segment a section is in, rather than using `segment_from_virtual_address`

Also, only run the function to remove these holes when actually signing/hashing/encrypting binaries, to prevent causing breakages to other commands like `uf2 convert` [... cut]
commit c291aeb2 (2025-04-22): Fix plugging alignment holes for encrypted binaries (#227)

Use new `segment_from_section` function, as virtual addresses are not unique per segment in encrypted binaries

Fixes #224
closed 2025-04-22 by will-v-pi