raspberrypi/picotool #225: Partition ID truncated to lower 32 bits only

richard9999999999, 2025-04-09:
I see following problem in all versions of Picotool including current DEVELOP.

My partition definition in json  file is as follows:
```
    {
		"name": "Wi-Fi Firmware",
		"id": "0x776966696669726d",
		"start": "3500k",
		"size": "240K",
		"families": ["data"],
		"permissions": {
		  "secure": "rw",
		  "nonsecure": "rw",
		  "bootloader": "rw"
		},
		"link": ["owner", 0],
		"ignored_during_riscv_boot": true,
		"no_reboot_on_uf2_download": true
	  },
```

however command:

`picotool info -all enc_bootloader.uf2`

shows following:
```
partition 4 (A ob/ 0):  0036b000->003a7000 S(rw) NSBOOT(rw) NS(rw), id=000000006669726d, "Wi-Fi Firmware", uf2 { data
                         }, arm_boot 1, riscv_boot 0
```
So partition ID is truncated to lower 32 bits only.

I use Mingw32 compiler on Windows to build the picotool..

will-v-pi, 2025-04-10:
I can reproduce this with prebuilt binaries on Windows - it seems the ID is getting truncated when creating the partitions from JSON, looks like a problem with parsing integer lengths

Could you try this diff and see if it fixes it for you?
```diff
diff --git a/cli.h b/cli.h
index 6db9b4f..97728cd 100644
--- a/cli.h
+++ b/cli.h
@@ -422,7 +422,7 @@ namespace cli {
         template<typename T>
         static std::string parse_string(std::string value, T& out) {
             size_t pos = 0;
-            long lvalue = std::numeric_limits<long>::max();
+            long long lvalue = std::numeric_limits<long long>::max();
             int64_t base = 10;
             if (value.find("0x") == 0) {
                 value = value.substr(2);

```

kilograham, 2025-04-10:
can we not use `uint64_t`

richard9999999999, 2025-04-11:
Yes, I confirm the problem is fixed after the patch above.
..the same with using uint64_t instead of long long type

closed 2025-04-11 by will-v-pi

lurch, 2025-04-15:
Looking at the function that gets modified in 450f8ed I see that it also does:
```c
            if (lvalue != (int64_t)lvalue) {
                return value + " is too big";
            }
```

Does that mean that picotool would also have problems displaying 64-bit partition IDs that are bigger than `int64_t` but smaller than `uint64_t` ? @will-v-pi

will-v-pi, 2025-04-15:
It doesn't appear to cause any issues (eg an ID of `0xDED3FFFF01234567` works fine) - note that this doesn't affect displaying partition IDs, only parsing JSON files to create partition tables

That check has been there since the original picotool (only modified in 2.0.0 to `int64_t` instead of `int`), so maybe worth pinging @kilograham to ask if it's still required?

lurch, 2025-04-15:
I can't really remember how C++ template programming works (been a long long time since I've done any of that), but looking at that function again, I wonder if both of the `int64_t` in [this section](https://github.com/raspberrypi/picotool/blob/master/cli.h#L448-L451):
```c
            if (lvalue != (int64_t)lvalue) {
                return value + " is too big";
            }
            out = (int64_t)lvalue;
```
ought to be `<T>` instead?

commit 450f8ed3 (2025-04-15): Fix partition ID truncation on Windows

Fixes #225