raspberrypi/picotool #251: Cannot build on windows with libusb

JohnAZoidberg, 2025-07-10:
I tried to do it myself and then I noticed the GitHub action CI also can't do it:

```
-- Could NOT find LIBUSB (missing: LIBUSB_LIBRARIES LIBUSB_INCLUDE_DIR) 
libUSB is not found - no USB support will be built
```

In the run of "[Build & Test (windows-latest, Ninja, mbedtls, libusb, compile)](https://github.com/raspberrypi/picotool/actions/runs/16102733671/job/45433976581#logs)"

lurch, 2025-07-11:
Well that's unexpected! @will-v-pi I guess the CI-test should check that in the libusb-included builds, the `picotool help` output lists the `load` command as being available?

lurch, 2025-08-27:
If you're unable to build `picotool` yourself, I believe the recommendation is to download it from https://github.com/raspberrypi/pico-sdk-tools/releases