raspberrypi/picotool #259: udev rules have wrong priority on Fedora

Windwoes, 2025-07-19:
The provided udev rules don't work on the fedora. The file needs to be renamed to `70-picotool.rules`.

lurch, 2025-07-20:
Could you provide some more info about why this is the case please? (i.e. why `99-picotool.rules` apparently doesn't work but `70-picotool.rules` does). Does the version of udev (or systemd?) in Fedora work differently to that in Debian?

Windwoes, 2025-07-20:
The description there about uaccess and plugdev is correct.

lurch, 2025-07-20:
> The description there about uaccess and plugdev is correct.

...in which case, your problem should already be solved on the `develop` branch of picotool? (by #230 )

will-v-pi, 2025-09-10:
This is fixed in master, as the file is now `60-picotool.rules`

closed 2025-09-10 by will-v-pi