raspberrypi/picotool #266: "permission failure" when trying to set page lock?

richard9999999999, 2025-08-03:
A fragment from my otp.json file which I am trying to load

```
    "PAGE53_LOCK1": "0x3d3d3d",
    "PAGE54_LOCK1": "0x3d3d3d",
    "PAGE55_LOCK1": "0x3d3d3d",
    "boot_flags1": {
        "key_valid": 1
    },
    "bootkey0": [
```

I am getting following error:

```
PAGE53_LOCK1:
  0x003d3d3d
ERROR: The RP2350 device returned an error: permission failure
```

I tried more pages (also not touched before)

Or did I miss something?

richard9999999999, 2025-08-03:
Now I tried page 1 and it works. Is there any reason it should not work for page 52 or 50?

will-v-pi, 2025-08-04:
See errata RP2350-E15 in the datasheet - the workaround is to use the `picotool otp permissions` command which loads a binary onto the chip to set the permissions.

closed 2025-08-04 by will-v-pi

richard9999999999, 2025-08-04:
Thank you, it is clear now.

richard9999999999, 2025-08-04:
Thank you. I think that message could save some time for developers like me :-)